hardbyte/python-can #1455: Negative values can't be sent within the can.Message type

dstruiksma, 2022-12-05:
### Describe the bug

When sending a message which requires a signed int the bytearray returns a value error because the byte must be in range(0,256)

### To Reproduce

`    driving_dynamics1 = can.Message(
            arbitration_id = 0x500, data=[
            np.uint8(12),
            np.uint8(20),
            np.int8(15),
            np.int8(-120),
            np.uint8(109),
            np.uint8(121),
            np.int8(-21),
            np.int8(0)
            ]
    )
`
Sending this message on a VCAN interface within linux returns a ValueError
### Expected behavior

I expected a way to configure the message for signed integers with a range of [-128 -> 127]

### Additional context

OS and version: Raspbian GNU/Linux 11 (bullseye
Python version: 3.9
python-can version: 4.4.0
python-can interface/s (if applicable): Virtual interface

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2022-12-06:
python-can sends and receive raw data. A [bytearray](https://docs.python.org/3/library/stdtypes.html#bytearray) cannot have negative values.

You need to encode your information into raw bytes before sending via CAN. On the receiving end you will have to decode this raw data to retrieve your information. Usually this is done with some CAN database format (.dbc, .arxml etc.).

dstruiksma, 2022-12-07:
> python-can sends and receive raw data. A [bytearray](https://docs.python.org/3/library/stdtypes.html#bytearray) cannot have negative values.
>
> You need to encode your information into raw bytes [...]

Thank you for the explanation I think I need to go another route for what I want to do.

closed 2022-12-07 by hardbyte