hardbyte/python-can #1461: Unable to Load PCANBasic Module on Windows on ARM

sanssecours, 2022-12-19:
### Describe the bug

Executing python-can code based on the PCANBasic API on Windows on ARM does not seem to work. The code line:

https://github.com/hardbyte/python-can/blob/f3136fb5e85f11ecb9ba83744a955a706f9dbde2/can/interfaces/pcan/basic.py#L661

is unable to load the DLL:

```
FileNotFoundError: Could not find module 'PCANBasic' (or one of its dependencies). Try using the full path with constructor syntax.
```

This error happens both on the x64 version of Python `3.10` and the (experimental) ARM version of Python `3.11`.

### To Reproduce

1. Install Windows on ARM (e.g. using [Parallels Desktop](https://www.parallels.com) on an Apple Silicon based Mac)
2. Install [PCAN driver](https://www.peak-system.com/quick/DrvSetup)
3. Install python-can:

    ```sh
    pip install python-can[pcan]
    ```

4. Run the following Python code:
   
   ```py
   from can.interfaces.pcan.basic import PCANBasic
   PCANBasic()
   ```

### Expected behavior

The code works without problems.

### Actual behavior

Execution fails with the following stack trace:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\rene\AppData\Local\Programs\Python\Python310\lib\site-packages\can\interfaces\pcan\basic.py", line 661, in __init__
    self.__m_dllBasic = windll.LoadLibrary("PCANBasic")
  File "C:\Users\rene\AppData\Local\Programs\Python\Python310\lib\ctypes\__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "C:\Users\rene\AppData\Local\Programs\Python\Python310\lib\ctypes\__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'PCANBasic' (or one of its dependencies). Try using the full path with constructor syntax.

Could not find module 'PCANBasic' (or one of its dependencies). Try using the full path with constructor syntax.
```

### Additional context

OS and version: Windows 11 on ARM 
Python version: Python 3.10.8
python-can version: python-can-4.0.0
python-can interface/s (if applicable): [PCAN-USB FD](https://www.peak-system.com/PCAN-USB-FD.365.0.html)

For now I use the following **(very ugly) workaround** to fix the problem:

```py
from re import match
# Loads the API on Windows
if match('^ARM', platform.processor()) and platform.machine() == 'AMD64':   
    # x64 emulation on ARM
    self.__m_dllBasic = windll.LoadLibrary("C:\\Program Files\\PEAK-System\\PEAK-Drivers 4\\APIs\\x64\\PCANBasic.dll")
else:
    # Native library on Windows and Windows on ARM
    self.__m_dllBasic = windll.LoadLibrary(find_library("PCANBasic"))
```

zariiii9003, 2022-12-21:
Not a bug, python just can't find the DLL. Add "C:\Program Files\PEAK-System\PEAK-Drivers 4\APIs\x64" to your `PATH` environment variable. You might have to restart your system afterwards.

sanssecours, 2022-12-21:
> Not a bug, python just can't find the DLL. Add "C:\Program Files\PEAK-System\PEAK-Drivers 4\APIs\x64" to your `PATH` environment variable. You might have to restart your system afterwards.

Thank you for the very helpful reply 👍. It seems the path for both the x64 and ARM version of the library were part of the system `PATH` variable already. I assume they were added by the [PCAN driver installer](https://www.peak-system.com/quick/DrvSetup). Anyway, python-can was still unable to find the DLL unless I added a call to `find_library`. Pull request #1463 contains a commit that adds this call.

commit e9252de7 (2022-12-21): PCAN: Fix detection of lib on Windows on ARM (#1463)

Before this commit detection of the PCAN DLL would fail on Windows on
ARM, regardless if you used the native Python version, or the x64
version of Python. After this fix detection should work properly as long
as the PCAN library for your version of Python is listed first in the
`PATH` variable. The default:

1. `C:\Program Files\PEAK-System\PEAK-Drivers 4\APIs\ARM64\` before
2. `C:\Program Files\PEAK-System\PEAK-Drivers 4\APIs\x64\`

should work if you use the (native) ARM version of Python. If you
reorder these paths, then loading the libr [... cut]
closed 2022-12-21 by zariiii9003