hardbyte/python-can #1473: generate .exe file with auto-py-to-exe tool unable to connect with can hardware

LucioMater, 2022-12-28:
Hello to everybody,   I have generated an exe file from a working .py script, but .exe can not connect to CAN BUS hardware. When I execute the .exe file  this exception is raised:
Exception: unable to connect to CAN bus, check hardware
[16612] Failed to execute script 'maincan' due to unhandled exception!

Thanks

OS and version: Windows 10
Python version:  3.10
python-can version: 4.0.0

zariiii9003, 2022-12-28:
No code example, no traceback, you don't even mention which interface you are using. 
Does it work without auto-py-to-exe?

LucioMater, 2022-12-28:
Yes the script works, i have problems only with executables

j-c-cook, 2022-12-28:
You likely need to look at the documentation of the executable creator package you are using. There are typically limited documentation on writing custom bindings. You may need to write a binding of your own. Python packages that create executables typically ship with the bindings for major packages (e.g. numpy, scipy, etc.). `python-can` is likely not included in their standard package.

LucioMater, 2022-12-29:
> You likely need to look at the documentation of the executable creator package you are using. There are typically limited documentation on writing custom bindings. You may need to write a binding of [...]

yes thanks with the folllowing command it works:
"pyinstaller --hidden-import=can.interfaces.pcan --onefile test.py"

closed 2022-12-29 by LucioMater