hardbyte/python-can #1487: Simple .log not working for WireShark

BenViete, 2023-01-16:
### Describe the bug
Since the update to version 4.1.0 the simple .log file cannot be loaded to Wireshark. If you try to import it generates stupid protocol errors, because the CAN protocol is not recognized.

### To Reproduce
Send a message on a CAN bus and save the received messages to a .log file. Try to open it with WireShark.

### Expected behavior
The .log file should be imported correctly and recognized as a CAN trace.

### Additional context

OS and version: MacOS Montery 12.6.2
Python version: PYthon 3.9.2
python-can version: 4.1.0
python-can interface/s (if applicable): SocketCAN

The problem is caused by adding the direction of the message to the log file in line 185 of canutils.py.

j-c-cook, 2023-01-20:
Are you saying the issue is the two following lines?
https://github.com/hardbyte/python-can/blob/69a5209edb893413d932d8881ad93b5cd599af95/can/io/canutils.py#L185-L186

BenViete, 2023-01-20:
No it's from the following two lines:
https://github.com/hardbyte/python-can/blob/69a5209edb893413d932d8881ad93b5cd599af95/can/io/canutils.py#L182-L183

j-c-cook, 2023-01-21:
@jazi007 @felixdivo The change in question was merged into master here: https://github.com/hardbyte/python-can/commit/4d9b32c99c333ce9a0cac89afa9aeb324b3b4f7f.

The associated PR is here: https://github.com/hardbyte/python-can/pull/1244.

hartkopp, 2023-01-21:
Hi @j-c-cook ,

the direction is an optional information that can be enabled e.g. in `candump`with the `-x` option:

```
user@linux:~$ candump -L -x any
(1674328177.352335) vcan0 123#112233 T
user@linux:~$ candump -L any
(1674328199.203737) vcan0 123#112233
user@linux:~$ candump -l -x any
Disabled standard output while logging.
Enabling Logfile 'candump-2023-01-21_201014.log'
user@linux:~$ cat candump-2023-01-21_201014.log 
(1674328218.255741) vcan0 123#112233 T
user@linux:~$
```

As you can see the optional `T` or `R` is placed after the correct CAN frame representation - to be compatible to previous log file parsers that should simply stop when the CAN frame (as they know it) is finished.

The question is what did the Wireshark parser do that it is not capable to cope with such an extension?!?

zariiii9003, 2023-01-22:
I suggest opening an issue at https://gitlab.com/wireshark/wireshark/-/issues
Since the python-can implementation is equivalent to can-utils  i will close this.

closed 2023-01-22 by zariiii9003

j-c-cook, 2023-01-22:
> I suggest opening an issue at https://gitlab.com/wireshark/wireshark/-/issues Since the python-can implementation is equivalent to can-utils i will close this.

@BenViete