hardbyte/python-can #1494: Vector 1610 wont connect, XL will

rackflot, 2023-01-23:
### Describe the bug

I started using the CanCaseXL channel 1 and it worked right away. I moved to a Vector 1630 using Channel 3. Now I get the following error for using channel 1 or 3. The mapping looks correct, is there an issue with the 1630?

Exception has occurred: VectorInitializationError
xlGetApplConfig failed (Vector HW Config: Channel '1' of application 'mf' is not assigned to any interface) [Error Code 255]
can.interfaces.vector.exceptions.VectorInitializationError: xlGetApplConfig failed (XL_ERROR) [Error Code 255]

During handling of the above exception, another exception occurred:

  File "D:\Users\mflood\OneDrive\CAN_Decoder\MFCan.py", line 78, in CAN_Driver
    self.CANbus = can.Bus(interface='vector',
  File "D:\Users\mflood\OneDrive\CAN_Decoder\Receive_all.py", line 76, in <module>
    MFCan.CAN_Driver()  #attach the vector driver
can.interfaces.vector.exceptions.VectorInitializationError: xlGetApplConfig failed (Vector HW Config: Channel '1' of application 'mf' is not assigned to any interface) [Error Code 255]
[image]
[image]

### To Reproduce

### Expected behavior

### Additional context

OS and version: win10, py38
Python version: 3.8
python-can version: 4.1
python-can interface/s (if applicable): Vector 1630A

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2023-01-23:
CAN 1 in Vector HW Conf is channel 0  in the vector xl driver. Channel 1 (CAN2) is not assigned as the error message says.

rackflot, 2023-01-25:
Thanks, looks like it takes it as an int or a str.
The other thing i did was downgrade back to 3.3.3. I think that the latest version has a problem with it too. I will experiement.

j-c-cook, 2023-01-31:
@rackflot Is this still an issue?

rackflot, 2023-02-01:
I am not sure. I went back to using my cancaseXL with only two channels on it. The 1630 has 4 CAN channels and provides logical channel allocation as per above CAN 1 channel 3. 
I also went back the previous version of python-can, 3.3.4. i have it all working now. I will double back and try my other can box before i update this yet. 

I will close this and pose the question again if I run into an issue. Thanks.

zariiii9003, 2023-02-02:
@rackflot You could call `can.detect_available_configs(interfaces=["vector"])` to find the correct channel index

rackflot, 2023-02-02:
well, I have found more detail on my issue. I have a working version of 3.3.3 on one pc. On the other pc I was trying to use 4.1.0, this was giving me the issues. I ried your detect call and it returned different values. I will try the latest knowing i can back install and continue forward here are some pictures.

[image]
[image]
[image]
[image]

rackflot, 2023-02-03:
i reopened with output from v3.3.3 and 4.1.0 and the error.

zariiii9003, 2023-02-03:
@rackflot v4.1.0 doesn't work with CancaseXL, see #1453. That is fixed in the develop branch already.

But your VN1610 and VN1630 should work without problems.