hardbyte/python-can #1495: No buffer space available

Hector897, 2023-01-23:
Hello, I am trying to make a can communication through the obd2 port of a vehicle through the raspberry pi and the pican2 duo. I am using a base code obtained from git hub which I leave below. The point is that at first when executing the code through the phyton3 obdii_logger.py command, I got an error in socketcan_native and extended_id. This was solved by replacing socketcan_native as socketcan and extended_id with is_extended_id. With this the problem is solved now the problem I have is with the buffer since when I execute the corrected code it gives me the error "No buffer space available". I was researching on the subject and I found a command which increases the size of the buffer which is "sudo ifconfig can0 txqueuelen 1000" when executing this command the error only changes the name since now it appears as "transmit buffer full". I need your help if someone went through the same thing.

Then I leave the code used or the url where they can find it
https://github.com/skpang/PiCAN-Python-examples/blob/master/obdii_logger.py

#!/usr/bin/python3
## obdii_logger.py
# This python3 program sends out OBDII request then logs the reply to the sd card.
# For use with PiCAN boards on the Raspberry Pi
# http://skpang.co.uk/catalog/pican2-canbus-board-for-raspberry-pi-2-p-1475.html
# Make sure Python-CAN is installed first http://skpang.co.uk/blog/archives/1220
#  24-08-16 SK Pang

import RPi.GPIO as GPIO
import can
import time
import os
import queue
from threading import Thread

led = 22
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(led,GPIO.OUT)
GPIO.output(led,True)

# For a list of PIDs visit https://en.wikipedia.org/wiki/OBD-II_PIDs
ENGINE_COOLANT_TEMP = 0x05
ENGINE_RPM          = 0x0C
VEHICLE_SPEED       = 0x0D
MAF_SENSOR          = 0x10
O2_VOLTAGE          = 0x14
THROTTLE            = 0x11

PID_REQUEST         = 0x7DF
PID_REPLY           = 0x7E8

outfile = open('log.txt','w')

print('\n\rCAN Rx test')
print('Bring up CAN0....')

# Bring up can0 interface at 500kbps
os.system("sudo /sbin/ip link set can0 up type can bitrate 500000")
time.sleep(0.1) 
print('Ready')

try:
    bus = can.interface.Bus(channel='can0', bustype='socketcan')
except OSError:
    print('Cannot find PiCAN board.')
    GPIO.output(led,False)
    exit()

def can_rx_task():  # Receive thread
    while True:
        message = bus.recv()
        if message.arbitration_id == PID_REPLY:
            q.put(message)          # Put message into queue

def can_tx_task():  # Transmit thread
    while True:

        GPIO.output(led,True)
        # Sent a Engine coolant temperature request
        msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_COOLANT_TEMP,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
        bus.send(msg)
        time.sleep(0.05)

        # Sent a Engine RPM request
        msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_RPM,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
        bus.send(msg)
        time.sleep(0.05)

        # Sent a Vehicle speed  request
        msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,VEHICLE_SPEED,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
        bus.send(msg)
        time.sleep(0.05)        

        # Sent a Throttle position request
        msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,THROTTLE,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
        bus.send(msg)
        time.sleep(0.05)
        
        GPIO.output(led,False)
        time.sleep(0.1)
                        
                        
q = queue.Queue()
rx = Thread(target = can_rx_task)  
rx.start()
tx = Thread(target = can_tx_task)
tx.start()

temperature = 0
rpm = 0
speed = 0
throttle = 0
c = ''
count = 0

# Main loop
try:
    while True:
        for i in range(4):
            while(q.empty() == True):   # Wait until there is a message
                pass
            message = q.get()

            c = '{0:f},{1:d},'.format(message.timestamp,count)
            if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_COOLANT_TEMP:
                temperature = message.data[3] - 40          #Convert data into temperature in degree C

            if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_RPM:
                rpm = round(((message.data[3]*256) + message.data[4])/4)    # Convert data to RPM

            if message.arbitration_id == PID_REPLY and message.data[2] == VEHICLE_SPEED:
                speed = message.data[3]                                     # Convert data to km

            if message.arbitration_id == PID_REPLY and message.data[2] == THROTTLE:
                throttle = round((message.data[3]*100)/255)                 # Conver data to %

        c += '{0:d},{1:d},{2:d},{3:d}'.format(temperature,rpm,speed,throttle)
        print('\r {} '.format(c))
        print(c,file = outfile) # Save data to file
        count += 1
            
except KeyboardInterrupt:
    #Catch keyboard interrupt
    GPIO.output(led,False)
    outfile.close()     # Close logger file
    os.system("sudo /sbin/ip link set can0 down")
    print('\n\rKeyboard interrtupt')

zariiii9003, 2023-01-23:
Is your interface connected to a bus with other active nodes?

hartkopp, 2023-01-23:
Hi @Hector897 ,

when you can send around 10 frames until it fails you have *very* likely a problem with your (physical) CAN bus setup!

You need

1.  termination resistors between CAN_L and CAN_H (2x 120 Ohms == 60 Ohms)
2. a second(!) CAN node with the same bitrate that does acknowledge the sent CAN frame

Hector897, 2023-01-23:
> ¿Su interfaz está conectada a un bus con otros nodos activos?

I am only activating can0 and through an obd2 cable I am connecting the port of the vehicle with the pican2 duo. In the vehicle cable I have already identified can h and can l.

Hector897, 2023-01-23:
> Hola @Hector897 ,
> 
> cuando puede enviar alrededor de 10 fotogramas hasta que falle, es _muy_ probable que tenga un problema con su configuración (física) del bus CAN.
> 
> Te hace falta
> 
> 1. resistencias de terminación entre CAN_L y CAN_H (2x 120 Ohmios == 60 Ohmios)
> 2. un segundo(!) Nodo CAN con la misma tasa de bits que reconoce la trama CAN enviada

The pican2 duo board already has the resistors integrated in the port can h and can l, I activate or deactivate them by means of a jumper. According to several sources of information, I understand that in the same way the obd2 port of the vehicle already has the resistance integrated in the can h and can l, the question is if it is necessary to activate the resistance of the pican2 plate at the moment of establishing a communication with the obd2 port? since in this case it would have two resistors activated, the one of the pican2 and the one of the obd2 port

hartkopp, 2023-01-23:
You should probably better start with connecting can0 to can1 and add the termination.

When this is set up correctly and you can send frames from can0 to can1 and vice versa you know that the CAN is working correctly.

Hector897, 2023-01-23:
> Probablemente debería comenzar conectando can0 a can1 y agregar la terminación.
> 
> Cuando esto está configurado correctamente y puede enviar fotogramas de can0 a can1 y viceversa, sabe que el CAN funciona correctamente.

I have already done it and if I can send messages from can0 to can1, for this I send messages on the can0 port with the command "cansend can1 001#1122334455667788" and on the can1 port I activate it as a receiver with the command "candump can1" and if I receive the messages sent. NOTE: IN ORDER TO ESTABLISH A COMMUNICATION CAN0 AND CAN1 I NEED TO ACTIVATE THE LOOPBACK

hartkopp, 2023-01-23:
No. When you connect can0 with can1 you can send CAN frames on can0 and you see it in candump on can0 and can1.

No loopback!! The loopback makes a shortcut inside the CAN controller and then it is not visible on the outside for ACK generation. Disable loopback on all CAN controllers.

Hector897, 2023-01-23:
To connect can0 with can1 it is necessary to have the 120 ohm resistors activated in each port, I mean having the resistor activated in the can0 port and the resistor activated in the can1 port

> No. Cuando conectas can0 con can1 puedes enviar fotogramas CAN en can0 y lo ves en candump en can0 y can1.
> 
> ¡¡Sin bucle invertido!! El bucle invertido crea un acceso directo dentro del controlador CAN y luego no es visible en el exterior para la generación de ACK. Desactive el bucle invertido en todos los controladores CAN.

To connect can0 with can1 it is necessary to have the 120 ohm resistors activated in each port, I mean having the resistor activated in the can0 port and the resistor activated in the can1 port

hartkopp, 2023-01-23:
Yes. When enabling the 120 Ohms on both ports this leads to 60 Ohms between CAN_L and CAN_H which is the desired termination.

Hector897, 2023-01-23:
> Sí. Al habilitar los 120 ohmios en ambos puertos, esto conduce a 60 ohmios entre CAN_L y CAN_H que es la terminación deseada.

In order to achieve can0 communication with can1, would it be to enable the can0 port and activate the can1 port as a receiver?

If I have two open windows of the raspberry terminal in a tab enable can0 as emitter
sudo /sbin/ip link set can0 up type can bitrate 500000

and in another tab enable can1 as receiver
candump can1

This is how communication is achieved between both linked ports?

hartkopp, 2023-01-23:
CAN interfaces are usually bi-directional.

```
sudo /sbin/ip link set can0 up type can bitrate 500000
sudo /sbin/ip link set can1 up type can bitrate 500000
```

In terminal1:
`candump any`

In terminal2:
`cansend can0 123#112233`

Look at terminal1:
```
user@raspi:~$ candump any
  can0  123   [3]  11 22 33
  can1  123   [3]  11 22 33
```

hartkopp, 2023-01-23:
Please check with `ip -det link show` if the loopback option is really disabled on both CAN interfaces.

Hector897, 2023-01-23:
> Compruebe si la opción de bucle invertido está realmente deshabilitada en ambas interfaces CAN.`ip -det link show`

When executing the command "ip -det link show", I get the answer "<loopback,up,lower_up>" I already deactivated the loopback but the loopback is not deactivated, what can I do?

I disable loopback mode with the following commands:
sudo ip link set can0 down
sudo ip link set can0 type can loopback off
sudo ip link set can up type can bitrate 500000

hartkopp, 2023-01-23:
Sure?

Can you please send the output of:

```
ip -det link show can0
ip -det link show can1
```

Hector897, 2023-01-23:
I have already deactivated the inverted loop, now I am carrying out the communication between can0 and can1, the question is why I do not receive any data when the two resistors are activated, it even appears "No buffer space available" and when I deactivate any of the two resistors I start to receive data and the "no buffer space available" message disappears

hartkopp, 2023-01-23:
Did you connect the two lines between can0 and can1?
Can you check the 60 Ohms with a multimeter?
And again ... send the output of:

ip -det link show can0
ip -det link show can1

Hector897, 2023-01-24:
> ¿Conectaste las dos líneas entre can0 y can1? ¿Puedes comprobar los 60 ohmios con un multímetro? Y otra vez... Envíe la salida de:
> 
> ip -det link show can0 ip -det link show can1

Sorry for the delay, And for this test I have connected can h of port can0 with can h of port can1 and the same for can l is connected to canl of the other port

I already measured the resistance and between can h and can l of port can0 there is a resistance of 60 ohms with the jumper on and if I remove the jumper I have a resistance of 120 ohms.

Hector897, 2023-01-24:
I have already done more tests and I can only receive data with candump any when the resistance is 120 ohms in can h and can l, and for this I have to deactivate either of the two resistors

hartkopp, 2023-01-24:
> I have already done more tests and I can only receive data with candump any when the resistance is 120 ohms in can h and can l, and for this I have to deactivate either of the two resistors

So can you finally see the traffic on can1 that you sent on can0 ?

Hector897, 2023-01-24:
Yes, I can now communicate and see the data traffic in can0 and can1. Only for this case I need to have a final resistance of 120 ohm since with 60 ohm I can't see the data traffic

Hector897, 2023-01-24:
i can see something like this

canbus1@raspberrypi:~$ candump any
  can0  123   [3]  11 22 33
  can1  123   [3]  11 22 33

hartkopp, 2023-01-24:
> i can see something like this
> 
> canbus1@raspberrypi:~$ candump any
> can0 123 [3] 11 22 33
> can1 123 [3] 11 22 33

That looks good.

Usually the CAN bus is very robust. Sometimes it also works with 120 Ohms or 30 Ohms. As 60 Ohms is the correct recommended termination: Have you really measured the termination with a multimeter or is this just an estimation according to the jumper settings?

Hector897, 2023-01-24:
> > Puedo ver algo como esto
> > canbus1@raspberrypi:~$ candump any
> > can0 123 [3] 11 22 33
> > can1 123 [3] 11 22 33
> 
> Eso se ve bien.
> 
> Por lo general, el bus CAN es muy robusto. A veces también funciona con 120 ohmios o 30 ohmios. Como 60 ohmios es la terminación recomendada correcta: ¿Realmente ha medido la terminación con un multímetro o es solo una estimación de acuerdo con la configuración del puente?

I understand, in my case it must be 120 ohms and you are correct I measured the resistance with a multimeter

Hector897, 2023-01-24:
Through the multimeter it tells me when it was 60 ohms and 120 ohms. Now to make the connection to the obd2 I only connect the port to the pican2 and execute the code or a configuration must be carried out

fabHO1, 2023-01-25:
I have tried everything that hartkopp indicates and it does not solve the problem when I test the python example that comes in the manual of the PICANDuo

  import can
  bus = can.interface.Bus(channel='can0', bustype='socketcan_native')
  msg = can.Message(arbitration_id=0x7de,data=[0, 25, 0, 1, 3, 1, 4, 1])
  bus.send(msg)

I don't get OSError: [Errno 105] No buffer space available but when I send the code that executes @Hector897 after a while I get the error.

Hector897, 2023-01-26:
Hello, to read the vehicle data through obd2 is it necessary to have can0 and can1 activated?        I realize that when I do dry tests and only activate can0 it automatically flags too many errors and when I activate both ports the errors do not appear.

hartkopp, 2023-01-26:
> Hello, to read the vehicle data through obd2 is it necessary to have can0 and can1 activated? I realize that when I do dry tests and only activate can0 it automatically flags too many errors and when I activate both ports the errors do not appear.

Maybe the two CAN controllers share the same interrupt line. And then the uninitialized CAN controller might do bad things on this interrupt line?!?!

hartkopp, 2023-01-26:
> import can bus = can.interface.Bus(channel='can0', bustype='socketcan') msg = can.Message(arbitration_id=0x7de,data=[0, 25, 0, 1, 3, 1, 4, 1]) bus.send(msg)

Have you ever tried to set the `dlc` value, e.g.

` msg = can.Message(arbitration_id=0x7de,dlc=8, data=[0, 25, 0, 1, 3, 1, 4, 1]) bus.send(msg)` 

I'm not sure how python handles this value when missing.

But when you still get such error "after a while", re-start here https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400850708

fabHO1, 2023-01-26:
> > import can bus = can.interface.Bus(channel='can0', bustype='socketcan') msg ​​= can.Message(arbitration_id=0x7de,data=[0, 25, 0, 1, 3, 1, 4, 1 ]) bus.enviar(mensaje)
> 
> ¿Alguna vez ha intentado establecer el `dlc`valor, por ejemplo,
> 
> ` msg = can.Message(arbitration_id=0x7de,dlc=8, data=[0, 25, 0, 1, 3, 1, 4, 1]) bus.send(msg)`
> 
> No estoy seguro de cómo Python maneja este valor cuando falta.
> 
> Pero cuando aún recibe ese error "después de un tiempo", reinicie aquí [# 1495 (comentario)](https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400850708)

I have not tried it I will try it thanks in case it works I will comment it.

fabHO1, 2023-01-26:
>>> import can 
>>> bus = can.interface.Bus(channel='can0', bustype='socketcan')
>>> msg = can.Message(arbitration_id=0x7de,dlc=8, data=[0, 25, 0, 1, 3, 1, 4, 1])
>>> bus.send(msg)
>>> notifier = can.Notifier(bus, [can.Printer()]

intente lo que me decias si habia probado, esta vez coloque una nueva linea notifier = can.Notifier(bus, [can.Printer()] la cual en el manual decia que era para recibir mensajes y obtube los siguientes resultados.

Timestamp: 1674766028.983273        ID: 0004      S  Rx  E                DL:     8       00  08  00  00  00  00  00  00     Channel: can0
>>> Timestamp: 1674766028.983273        ID: 0004      S  Rx  E                DL:     8       00  08  00  00  00  00  00  00     Channel: can0

hartkopp, 2023-01-26:
https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400983100

Hector897, 2023-02-10:
I appreciate your help, it was very important thanks to this and I solved my problem thank you very much. 🥇

closed 2023-02-10 by Hector897