hardbyte/python-can #1513: question: decoding large J1939 messages

acode-x, 2023-01-28:
Hi,

I have a query regarding handling J1939 messages having length > 8 bytes.
I am using https://github.com/nberlette/canbus/blob/main/dbc/j1939.dbc as a reference.
It has messages:
```bash
BO_ 2633939198 GTFI: 22 Vector__XXX
 SG_ ...

BA_ "GenMsgSendType" BO_ 2633939198 8;
BA_ "VFrameFormat" BO_ 2633939198 3;

BO_ 2633832958 ILI: 200 Vector__XXX
 SG_ ...
 
BA_ "VFrameFormat" BO_ 2633832958 3;
```

I have used python-can to decode standard messages of length 8 bytes, but not sure how to read and process length 22 / 200.
Requesting guidance on same.

Thanks

j-c-cook, 2023-01-28:
I am familiar with a CAN base frame format and an extended frame format (see ISO 11783 part 3). Both of those formats have a 64 bit data field. Can a message with more than 64 bits be sent, or do you have to send multiple messages?

zariiii9003, 2023-01-28:
J1939 defines a transport protocol, so you don't really need to work on raw CAN message level.
Check https://github.com/juergenH87/python-can-j1939 or https://github.com/danielhrisca/asammdf if you can log to mf4 format.

acode-x, 2023-01-31:
I was referring to multi frame (TP.CM and TP.DT)
Thanks for sharing above link python-can-j1939.

closed 2023-01-31 by acode-x