hardbyte/python-can #1523: TRC file cannot be opened to write on it. (TypeError)

alexamatausa, 2023-02-16:
When trying to create a Logger with a file in TRC format, the following error occurs: 

```
Traceback (most recent call last):
  File "c:\Users\aamat\OneDrive - AUSA CENTER S.L.U\Documentos\Programa de captura de PCAN\main.py", line 26, in <module>
    trc = can.Logger(f.name)
          ^^^^^^^^^^^^^^^^^^
  File "C:\Users\aamat\AppData\Local\Programs\Python\Python311\Lib\site-packages\can\io\logger.py", line 95, in __new__
    raise ValueError(
ValueError: No write support for this unknown log format ".trc"
```
Instead, by running the same program code and using the .log or .csv extension. I do get the output file. Is it possible that it is a bug?

I have tried creating a PathLib.PureWindowsPath class object instead of using a str to reference the object.

Please find attached the code and modules I have used to detect this problem.

[cantotrcfile.zip](https://github.com/hardbyte/python-can/files/10756987/cantotrcfile.zip)

zariiii9003, 2023-02-24:
I would be great if you used the bug report template. Which version are you using?

alexamatausa, 2023-02-24:
I'm using the version 4.0.0 of the library.

alexamatausa, 2023-02-24:
I'm using the version 4.0.0 of the library. I have seen that in version 4.0.0 of the library the TRC format is not available. I will do an update of the library.

closed 2023-02-24 by alexamatausa