hardbyte/python-can #1524: hi,there is something wrong when i use your examples which name is asyncio_demo.py, i changed interface from virtual to pcan, and changed channel  to PCAN_USBBUS1, but when i run it,the log is not same as the id i send

Wys295723, 2023-02-18:
def print_message(msg: can.Message) -> None:
    """Regular callback function. Can also be a coroutine."""
    print(msg)

async def main() -> None:
    """The main function that runs in the loop."""

    with can.Bus(
        interface="pcan", channel="PCAN_USBBUS1", receive_own_messages=True
    ) as bus:
        reader = can.AsyncBufferedReader()
        logger = can.Logger("logfile.asc")

        listeners: List[MessageRecipient] = [
            print_message,  # Callback function
            reader,  # AsyncBufferedReader() listener
            logger,  # Regular Listener object
        ]
        # Create Notifier with an explicit loop to use for scheduling of callbacks
        loop = asyncio.get_running_loop()
        notifier = can.Notifier(bus, listeners, loop=loop)
        # Start sending first message
        bus.send(can.Message(arbitration_id=0, data=[11,22,33,44]))

        print("Bouncing 10 messages...")
        for _ in range(10):
            # Wait for next message from AsyncBufferedReader
            msg = await reader.get_message()
            # Delay response
            await asyncio.sleep(0.5)
            msg.arbitration_id += 1
            bus.send(msg)

        # Wait for last message to arrive
        await reader.get_message()
        print("Done!")

        # Clean-up
        notifier.stop()

if __name__ == "__main__":
    asyncio.run(main())

result:
Timestamp: 1676724128.607573        ID: 0001    S Rx E              DL:  4    00 02 00 00
Timestamp: 1676724128.607621        ID: 0001    S Rx E              DL:  4    00 02 00 00
Timestamp: 1676724128.607669        ID: 0001    S Rx E              DL:  4    00 02 00 00
Timestamp: 1676724128.607717        ID: 0001    S Rx E              DL:  4    00 02 00 00
Timestamp: 1676724128.607765        ID: 0001    S Rx E              DL:  4    00 02 00 00
Timestamp: 1676724128.607813        ID: 0001    S Rx E              DL:  4    00 02 00 00
...

Wys295723, 2023-02-18:
i realy want konw  if it wroks when i set receive_own_messages True

catmemo, 2023-03-29:
> : 0001 S

It seems that, PCAN has not the parameter - receive_own_messages.