hardbyte/python-can #1533: can.interfaces.serial conflicts with pyserial

RyanHope, 2023-03-06:
pyserial provides the serial package which has a namespace conflict in the seeedstudio bus (and possibly other interfaces using pyserial) with the can.interfaces.serial package

should probably rename can.interfaces.serial to can.interfaces.serial_can

zariiii9003, 2023-03-06:
That's a breaking change. Can you provide an example where an aliased import can not solve the problem?

RyanHope, 2023-03-06:
> That's a breaking change. Can you provide an example where an aliased import can not solve the problem?

I guess I don't care how its fixed, but interfaces are currently broken.

zariiii9003, 2023-03-06:
> I guess I don't care how its fixed, but interfaces are currently broken.

Could you share the traceback?

RyanHope, 2023-03-07:
> > I guess I don't care how its fixed, but interfaces are currently broken.
> 
> Could you share the traceback?

instantiate a seeedstudio bus, it will fail because serial.Serial does not exist

RyanHope, 2023-03-07:
Im gonna close this for now because I can never get inbranch version of the seeedstudio interface to work. I have to rely on my own implementation and that seems to be working right now without renaming any branches.

closed 2023-03-07 by RyanHope

zariiii9003, 2023-03-07:
> instantiate a seeedstudio bus, it will fail because serial.Serial does not exist

Did you install pyserial? I cannot reproduce your error. If it was broken, i'm sure we would get complaints about it.