hardbyte/python-can #1537: Kvaser cannot send frames

bessman, 2023-03-09:
### Describe the bug
This
```python
bus = can.Bus(interface="kvaser", channel=0, ...)
bus.send(can.Message())
```
causes a bus error (hardware error led flashes). No traceback or the like; only indication in software that something is wrong is that error frames start coming in.

Receiving frames works.

This can be worked around by either:
1. Setting `single_handle=True`.
or
2. First initializing the bus using some other software, such as Kvaser CanKing (after which `single_handle=False` also works).

### To Reproduce
See above.

### Expected behavior
It should be possible to send frames using kvaser driver without the above workarounds.

### Additional context

OS and version: Windows 10
Python version: 3.10.10
python-can version: 4.1.0
python-can interface/s (if applicable): "kvaser", w. Kvaser Blackbird v2 hardware.

zariiii9003, 2023-03-09:
I wonder why the KvaserBus uses two handles by default. I'd suggest to always use just one handle for read and write and configure it for non-exclusive access. The tx echo can be configured by `receive_own_messages`.

bessman, 2023-03-09:
> The tx echo can be configured by `receive_own_messages`.

Only if `single_handle=False`. If `single_handle=True`,
- messages sent by the same `KvaserBus` instance cannot be echoed, and
- echo of messages sent by other applications on the same channel cannot be disabled.

Which is why `single_handle=False` is the default, I would guess.

bessman, 2023-03-09:
I haven't used python-can on Windows in a long time, but I'm pretty sure this used to work back in... 2019 or so? I'll try and see if it works with python-can ~3.3, and if so bisect the breakage.

It also seems like something that someone should have noticed if it's been broken for a while, so I'm curious it's something related to my setup. I've reproduced the problem across multiple computers, but since I'm in a corporate setting they all have more or less the same configuration.

zariiii9003, 2023-03-09:
> Only if `single_handle=False`. If `single_handle=True`,
>
> * messages sent by the same `KvaserBus` instance cannot be echoed, and
> * echo of messages sent by other applications on the same channel [...]

I cannot find this limitation in the Kvaser API documentation 🤔

bessman, 2023-03-09:
I'm going off `KvaserBus`'s docstring:
>  :param bool receive_own_messages:
>  If messages transmitted should also be received back.
>  Only works if single_handle is also False.
>  If you want to receive messages from other applications on the same
>  computer, set this to True or set single_handle to True.

I'll do some testing.

bessman, 2023-03-10:
> If `single_handle=True`,
> * messages sent by the same `KvaserBus` instance cannot be echoed, and
> * echo of messages sent by other applications on the same channel cannot be disabled.

After testing, I can confirm this behavior. Whether this is a limitation in Kvaser CANlib or in the `KvaserBus` implementation I do not know.

Additionally, further testing revealed that reception of frames is also spotty when `single_handle=False`. Some frames are received, others are not.

bessman, 2023-03-10:
This is a limitation of CANLib:
> * messages sent by the same `KvaserBus` instance cannot be echoed

A CANLib handle cannot receive its own transmitted messages; `canIOCTL_SET_LOCAL_TXECHO` only controls whether the handle receives messages transmitted by *other* handles using the same channel. 

This is due to how `KvaserBus` is implemented:
> * echo of messages sent by other applications on the same channel cannot be disabled.

https://github.com/hardbyte/python-can/blob/40968170de9ff909f401fb38a9978de162379692/can/interfaces/kvaser/canlib.py#L478