hardbyte/python-can #1547: Package not re-exporting symbols in package __init__.py

tjohanssonn, 2023-03-24:
### Describe the bug
`python-can` is not re-exporting symbols correctly. This causes Pylance to not provide Intellisense when using e.g. Bus class, and also warns about usage of private import:

`"Bus" is not exported from module "can" Pylance[reportPrivateImportUsage]`

Link to webpage: [reportPrivateImportUsage](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportPrivateImportUsage)

### To Reproduce

1. Editor: Visual Studio Code
2. Language server: Pylance
3. Set configuration `"python.analysis.typeCheckingMode": "basic"`
4. Minimum code example:
```python
import can

bus = can.Bus(bustype='socketcan', channel='vcan0', bitrate=500000)
```

### Expected behavior

Symbols imported in package/__init__.py should be publicly available, and not trigger a private import usage warning.

For more context on Pylance stance on this issue, see e.g. <https://github.com/microsoft/pylance-release/issues/2953#issuecomment-1168408943>
### Additional context

OS and version: Windows 10 22H2 OS build 19045.2728
Python version: 3.10.2
python-can version: 4.1.0
python-can interface/s (if applicable):

### Screenshot of warning(s)
[image]

felixdivo, 2023-03-30:
Interesting, that this did not come up earlier. Would setting `__all__` solve this?

It seems to me like we probably want that attribute anyways.

zariiii9003, 2023-03-30:
> Interesting, that this did not come up earlier. Would setting `__all__` solve this?
> 
> It seems to me like we probably want that attribute anyways.

I'd prefer the `import x as x` method. With `__all__` you would always need to make changes in two places.

felixdivo, 2023-03-30:
I see that, but by adding the `__all__`, we'd kill two birds with one stone: Pleasing the type cheker + making `from can import *` do something more controlled.

closed 2023-04-01 by zariiii9003

tjohanssonn, 2023-04-04:
Tested branch and it seems to be working, thank you for the fix 👍

PR #1551 (2023-04-01): Export symbols to satisfy type checkers
Closes #1547 

This also introduces ruff with isort, pyupgrade and the unused-imports check.

@felixdivo The version with the `import x as x` reexport can be seen [here](https://github.com/zariiii9003/python-can/blob/81a610820fa34c71461e6027334193b55c50043e/can/__init__.py). I think that one is less ugly, but i don't care too much.