hardbyte/python-can #1548: 为同一个设备的不同通道设置波特率

Memo654, 2023-03-28:
### Is your feature request related to a problem? Please describe.

目前，Bus类的接口中channel支持list/tuple类型，但bitrate类型为int， 这样就会造成打开多个通道时，所有通道的波特率相同。
我想对同一个Bus的不同的通道配置不同的波特率
[image]
### Describe the solution you'd like

我认为bitrate接口的数据类型可以扩展支持list/tuple类型，并于channel对齐
### Describe alternatives you've considered

无
### Additional context

无

### Note
我没有检索到有类似的问题，如果有重复，请告诉我，我会关闭这个问题。

Memo654, 2023-03-31:
@zariiii9003 Ok, I update the comment.

zariiii9003, 2023-03-31:
You could create multiple `CanalystIIBus` instances with different bitrates. While i see, that it could be useful to accept Sequences for bitrate, that would be quite a big change.

Memo654, 2023-04-03:
Yeah, It looks like a big chanage. I will change some code when I go off work. Hope it will be helpful in the future.