hardbyte/python-can #1581: can.BLFReader: Stuck printing "Unknown object header version (0)", cannot error handle.

desmotronic, 2023-04-29:
### Describe the bug
When reading certain BLF files that may be corrupt, the can.BLFReader() function gets stuck with the terminal output continuously printing "Unknown object header version (0)".  Because this does not raise an error flag, there is no way to get out of this loop using traditional error handling methods.  It seems maybe modifying the following section of can.io.blf would prevent the script from getting stuck here?

`else:
                LOG.warning("Unknown object header version (%d)", header_version)
                pos = next_pos
                continue`

To:
`else:
                raise BLFParseError("Unknown object header version (%d)", header_version) from None

### To Reproduce
I am not sure what is wrong with the particular BLF file this is getting stuck in but I cannot share it unfortunately.

### Expected behavior
If "Unknown Object header Version (0)" is encountered, raise an exception flag so that we can stop processing the faulted file.

### Additional context

OS and version: Windows 10
Python version: 3.11.2
python-can version: 4.1.0

zariiii9003, 2023-04-30:
Create a PR with your suggested changes and if you are lucky, @christiansandberg might take a look at it 👀

christiansandberg, 2023-04-30:
My thought behind it was that maybe there is just one or two of these occurrences in a file so maybe you don't want to bail out if the rest of the file is fine. But I guess it's more likely that either there are no unknown headers or the file will be completely unreadable anyway.