hardbyte/python-can #1582: Not able to transmit more than 8bytes single frame in can bus using python

FarhatJ, 2023-04-29:
### Describe the bug
I am trying to transmit more than 8 bytes of single frame size(CAN FD) for classci CAN channel using can.Message.
Also using Vector Canoe Application on Windows.

### To Reproduce

```python
with concurrent.futures.ThreadPoolExecutor() as executor:
            future = executor.submit(self.listen_for_port, port=port)

            msg0 = can.Message(arbitration_id=0x580, dlc=15, is_fd=True,
                               is_extended_id=True, bitrate_switch=True,
                               data=[3, 3, 3, 3, 3, 3, 3, 3, 3, 3])

            msg1 = can.Message(arbitration_id=0x582, dlc=15, is_fd=True,
                               is_extended_id=True, bitrate_switch=True,
                               data=[3, 3, 3, 3, 3, 3, 3, 3, 3, 3])

            messages = [msg0, msg1]
            vbus1.send_messages(messages, 1000)
            time.sleep(2)
            udp_frames = future.result()
```
I am getting below error when data is more than 8.

[image]

When data is 8 or less than 8, transmission works

[image]

### Expected behavior
Should be able to see more tahn 8 bytes of data on canoe can channel when data=[3, 3, 3, 3, 3, 3, 3, 3, 3, 3]

### Additional context

OS and version: Windows
Python version: 3.7.9
python-can version: 4.2.0
python-can interface/s (if applicable):

I though initially the issue is on isotp protocol so raised issue in  Python-can-isotp repository but after few discussion suggestion was to raise issue in Python-can layer
For more detail: https://github.com/pylessard/python-can-isotp/issues/85
</details>

zariiii9003, 2023-04-29:
CAN 2.0 can not transmit more than 8 bytes per message. CAN FD can transmit up to 64 bytes.

FarhatJ, 2023-04-29:
I have initalised fd=True:
My can bus obj looks like this:

```python
self.bus = can.Bus(channel=net_channel,
                           interface='vector',
                           receive_own_messages=True,
                           app_name='CANoe',
                           fd=True,
                           bitrate=500000)
```
Sending 2 different length of data:

data1=[3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3])
data2=[3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3])
DLC does vary first one is 14 and second one is 10 but i still do not see full data on single frame.

[image]

I even configured my can channel for CAN FD
[image]

Is there anything else that I am missing?

zariiii9003, 2023-04-30:
According to your traceback, you did NOT set `fd=True` when you created the issue.

When you are connected, do you see two baud rates here or just one?
[image]

FarhatJ, 2023-04-30:
@zariiii9003  yes you are correct, initially fd=True was not set, after your information I did set.

When I am connected to canoe application, I do see 2 Baud rates.
[image]

FarhatJ, 2023-04-30:
I found the issue why I am not able to see the full data on CAN Channel.
For isotp protocol to get full data of 64 bytes on single frame, parameter that i have set are:
1. tx_data_length: 64,
2. tx_data_min_length: 64,      # this make sure my data length is 64, and dlc varies depending on the data i am sending

Is there any related param for **tx_data_min_length** in can.Message() to be set other than dlc??