hardbyte/python-can #1588: [question] possible cause for vector XL_ERR_INVALID_HANDLE

danielhrisca, 2023-05-10:
### Describe the bug
I setup several buses on a VX1135 device
[image]

Each bus is being handled in a dedicated Python process.
I am able to start the buses and acquire CAN data, but when I want to shutdown the buses I often get this error:

```
Traceback (most recent call last):
  File "d:\daxil\xil\devices\can_device.py", line 1727, in _acquire
  File "d:\Tools\_venv\pyvenv_daxil2_Python310_x64\lib\site-packages\can\interfaces\vector\canlib.py", line 629, in shutdown
  File "d:\Tools\_venv\pyvenv_daxil2_Python310_x64\lib\site-packages\can\interfaces\vector\xldriver.py", line 37, in check_status_operation
can.interfaces.vector.exceptions.VectorOperationError: xlDeactivateChannel failed (XL_ERR_INVALID_HANDLE) [Error Code 155]
```
[image]

Do you know what can cause this?

zariiii9003, 2023-05-10:
I think XL_ERR_INVALID_HANDLE occurs, when the vector device is disconnected physically.

danielhrisca, 2023-05-10:
Yes, the bus is completely handled in the dedicated process and the problem occurs on the shutdown call. 
The VX is not disconnected, I also have a XCP connection with the VX that is running in parallel with no issues.

zariiii9003, 2023-05-11:
Did you try 4.2? If so, did it help?

Prindl, 2023-06-14:
I'm thinking the reason for your error during the shutdown might be that you are trying to close an already closed channel. Are you really using separate [processes](https://docs.python.org/3/library/multiprocessing.html) or are you using [threads](https://docs.python.org/3/library/threading.html)? I remember getting an error from the XL Driver Library when a thread tried to deactivate an already deactivated channel. Unless you provide more of your source it is almost impossible to tell what the cause of your issue is...

Wouldn't a possible solution be to check for an invalid port handle before calling the xlDeactivateChannel in [canlib.py](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/vector/canlib.py).

danielhrisca, 2023-07-05:
I'm using `multiprocessing.Process` not `threading.Thread`. In the same function the bus is created, then the code runs a `while` loop gathering messages from the bus until a stop variable is set, and finally the bus is shut down .

I've gotten more logs from users and now I see additionally this error:

[image]

Prindl, 2023-07-08:
So the function that you are passing to the `multiprocessing.Process` object creates the instance of the `VectorBus`? In that case I don't see why it should not work... However, when you check the documentation of the xlDriverLibrary it says the following:

```
XL_ERR_PORT_IS_OFFLINE: The user called a port function whose execution must be online, but the port is offline.
XL_ERR_INVALID_HANDLE: The user passed an invalid handle.
```

That means somewhere in your code you try to access the bus after the port was closed(`xlClosePort `is called in `shutdown`). Additionally, you seem to be trying to deactivate the channel after the port was closed and hence you get the `XL_ERR_INVALID_HANDLE`. All this leads me to believe, that you share a bus instance(or in terms of the xl driver library a port) between multiple threads/processes. Again, without any code it's hard to pinpoint the error...