hardbyte/python-can #1589: timestamp pcan incorrect, vector correct

skipper85, 2023-05-11:
### Describe the bug
One hour difference in timestamp between pcan and vector dongle.
The vector dongle returns correct timestamp.

pcan timestamp: 1683809683.251153
vector timestamp: 1683806083.2688096

### To Reproduce
make two Bus interfaces, one with vector VN1610 and other one with Peak PCAN-USB FD

### Expected behavior
Correct timestamp from the pcan dongle

### Additional context

OS and version: Windows 10
Python version: 3.10.8
python-can version: 4.1.0
uptime: 3.0.1
python-can interface/s (if applicable): Vector VN1610, Peak PCAN-USB FD
PCANBasic.dll : 4.6.0.600

<details><summary>Traceback and logs</summary>

```python

import can
from pprint import pprint

def test_pcan():
    bus = can.interface.Bus(
        bustype='pcan',
        channel='PCAN_USBBUS1',
        fd=True,
        f_clock_mhz=80,
        nom_brp=1,
        nom_tseg1=127,
        nom_tseg2=32,
        nom_sjw=40,
        data_brp=1,
        data_tseg1=9,
        data_tseg2=6,
        data_sjw=4)
    
    msg = bus.recv(timeout=5)
    print(f"pcan timestamp: {msg.timestamp}")

def test_vector():
    bus = can.interface.Bus(
        bustype='vector',
        channel=0,
        fd=True,
        bitrate=500_000,
        data_bitrate=5_000_000,
        sjw_abr=32,
        tseg1_abr=127,
        tseg2_abr=32,
        sjw_dbr=6,
        tseg1_dbr=9,
        tseg2_dbr=6)
    
    msg = bus.recv(timeout=5)
    print(f"vector timestamp: {msg.timestamp}")

test_pcan()
test_vector()

```
</details>

zariiii9003, 2023-05-11:
You probably need the `uptime` library. It is automatically installed with `pip install python-can[pcan]`

skipper85, 2023-05-12:
Hi,

Thank you for your fast response.
I tried the following

``` powershell
python -m venv env
.\env\Scripts\Activate.ps1
pip install python-can[pcan]

pip list
Package           Version
----------------- -------
packaging         23.1
pip               22.2.2
python-can        4.2.0
pywin32           306
setuptools        63.2.0
typing_extensions 4.5.0
uptime            3.0.1
wrapt             1.15.0
```

I ran the code again and there is still an hour difference in the timestamp of the pcan dongle.

pcan timestamp:   1683875166.8466148
vector timestamp: 1683871566.9361348

simeon-s1, 2023-08-21:
hey @skipper85 we're encountering a similar problem, did you eventually fix it?

skipper85, 2023-08-24:
hello @simeon-s1 , I did not fix it yet, still need to dive into it.

zariiii9003, 2023-08-25:
try #1651

skipper85, 2023-09-08:
Hello @zariiii9003 ,

Thank you for the pull request.

After installing with

``` shell
python -m venv env

git clone https://github.com/zariiii9003/python-can.git
git switch pcan_timestamp
cd python-can

python -m pip install --user pipx

python -m pipx run build
python -m pipx run twine check dist/*

..\env\Scripts\Activate.ps1

python -m pip install -e .

python -m pip install uptime
```
the timestamp is ok.

``` python

import can
import time

def test_pcan():
    bus = can.interface.Bus(
        bustype='pcan',
        channel='PCAN_USBBUS1',
        fd=True,
        f_clock_mhz=80,
        nom_brp=1,
        nom_tseg1=127,
        nom_tseg2=32,
        nom_sjw=40,
        data_brp=1,
        data_tseg1=9,
        data_tseg2=6,
        data_sjw=4)
    
    msg = bus.recv(timeout=5)
    print(f"pcan timestamp: {msg.timestamp}")
    print(f"timestamp: {time.time()}")

test_pcan()

```

pcan timestamp : 1694172746.938586
timestamp : 1694172746.9290798

zariiii9003, 2023-09-08:
Thanks for the feedback

PR #1651 (2023-09-08): Fix PCAN timestamp
Closes #1589
closed 2023-09-08 by zariiii9003