hardbyte/python-can #1591: Can't open CAN Bus for Vector Device interface.

nateimig18, 2023-05-12:
### Describe the bug
VectorBus has default app_name "CANalyzer" causing issue finding channels that haven't been associated yet.

### To Reproduce
Install drivers for the device first.
Connect a Vector VN1630 device or similar.
'''
import can

bus = can.Bus(bustype='vector', channel=0, bitrate=500000)
# can.interfaces.vector.exceptions.VectorInitializationError: xlGetChannelIndex failed (XL_ERR_HW_NOT_PRESENT) [Error Code 129]
# python-can version 4.2
'''

### Expected behavior
No errors on runtime 

### Additional context

OS and version: Windows
python-can version: 4.2
python-can interface/s (if applicable):

zhangshengping, 2023-05-24:
I got the same execption raise!

YongjinZhang-Eaton, 2024-02-01:
it does work now?
I got the same problem

StarsLighttheWay, 2024-08-18:
Excuse me, is there a solution now