hardbyte/python-can #1600: How do I use TRC file version 1.1? default it's using 2.1.

ArcticBeat05, 2023-05-19:
### Describe the bug
If i modify my code to use     logger = TRCWriter(f'{folder_path}_can_log_{now}.trc', file_version=TRCFileVersion.V1_1), i get an error. I suppose I am calling the version incorrectly or it might be a bug. 

### To Reproduce
This is what my current code looks like, which is running fine. 
`# Create a logger object with the filename and any otherarguments
def create_logger():
    now = datetime.datetime.now().strftime("%Y-%m-%d_%H-%M-%S")
    logger = TRCWriter(f'{folder_path}_can_log_{now}.trc') 
    return logger

def log_can_messages(): # define a function for logging
    try:
        logger = create_logger() # create a new logger
        start_time = time.time() # record the start time
        for msg in can_bus: # or rename can_bus to bus
            logger.on_message_received(msg)
            #print(msg)
            elapsed_time = time.time() - start_time # check the elapsed time
            if elapsed_time >= 60: # if 1 minutes have passed
                logger.stop() # stop the current logger
                logger = create_logger() # create a new logger
                start_time = time.time() # reset the start time
    finally:
        logger.stop()`

### Expected behavior

### Additional context

OS and version: Windows 10
Python version: 3.11.1
python-can version: 4.2.1
python-can interface/s (if applicable): PCAN

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

lumagi, 2023-06-09:
Hi @ArcticBeat05,
Please provide an exception trace of your error.

pkess, 2023-09-14:
Hi @ArcticBeat05 ,
i wrote the first version of the trc file reader and writer.
The way you call this is basically correct, but switching the file version was never implemented i think. I prepared it but there will be some work to support it. Is there any specific need for the 1.1 version or can you handle with the 2.1 as well?