hardbyte/python-can #1605: Ixxat USB-to-CAN v2 | python-can: Close bus, ACK and send_periodic errors

AntoineBricard, 2023-05-26:
Hi,

I'm new with python-can (package version= 4.2.1, running on python3.9.6, under Windows 10) and I wanted to perform a test to send/read CAN message using USB-to-CAN v2 (and SimplyCAN as target).
But I faced few issues using python-can:

**1. Close bus error:**

```
import can
bus = can.interface.Bus(interface='ixxat', channel=0, bitrate=250000, extended=True)
heartbeat = can.Message(arbitration_id=0x11487020, is_extended_id=True, data=[0x01])
bus.send(heartbeat)
bus.shutdown()
```

Above is the code I used to send CAN message. The point is once the code has been performed, it seems to not close properly the CAN bus. 
Bellow the error I got:
```
IXXATBus was not properly shut down
Exception ignored in: <function BusABC.__del__ at 0x000001ACF1418CA0>
```

Is there a way to properly close the CAN bus ? As i can rerun this script even after the error, should I just not taking account this error ?

**2. send_periodic :**
I wanted to send CAN message every seconds using the method "send_periodic" but it raise the bellow exception:
`AttributeError: 'IXXATBus' object has no attribute '_periodic_tasks'`

As in the doc, it's written that for Ixxat devices, the method "send_periodic" is supported, what did I misunderstand for the use of send_periodic method using Ixxat devices ? Bellow the code for the periodic message:
```
import can
bus = can.interface.Bus(interface='ixxat', channel=1, bitrate=250000, extended=True)
heartbeat = can.Message(arbitration_id=0x11487020, is_extended_id=True, data=[0x01])
bus.send_periodic(heartbeat, period=1)
bus.shutdown()
```

**3. NACK error :**

If the target has a problem and couldn't ACK the CAN message, the python-can API print errors logs as bellow:
```
CAN acknowledgment error
CAN message flags bAddFlags/bFlags2 0x00 bflags 0x05
```

And even once the target is reconnected this error message still appears. Is there a way to avoid them ?

Thanks in advance for any help,

Best regards,
Antoine

zariiii9003, 2023-05-26:
Hi, thanks for reporting
1. Try #1606 
2. The traceback might be helpful
3. 🤷‍♀️

AntoineBricard, 2023-05-30:
Hi zariiii9003,

1. Thanks for the link, using it could avoid the incorrect shutdown issue.

2. Here is the log I got:
```
  File "C:\Travail\4MOD - Test\Test_USB_to_CAN\test.py", line 8, in <module>
    bus.send_periodic(heartbeat, period=1, duration=10)
  File "c:\Travail\4MOD - Test\Test_USB_to_CAN\.venv\lib\site-packages\can\bus.py", line 245, in send_periodic
    periodic_tasks = self._periodic_tasks
AttributeError: 'IXXATBus' object has no attribute '_periodic_tasks'
```

But using a SimplyCAN as target, I could see that the `bus.send_periodic` method is working but it raise the above exception.
Is there a way to fix it or should I just use try/except statement in order to skip the Exception ?

Best regards,
Antoine

zariiii9003, 2023-07-06:
@AntoineBricard Can you test #1628?

AntoineBricard, 2023-07-07:
Hi zariiii9003,

Thanks for the info. I will tests when possible but it will not be before a month.

Best regards,
Antoine Bricard