hardbyte/python-can #1608: AttributeError: 'Channel' object has no attribute 'attachments' ==> Bug when I call can.MF4Writer()

henrique-rsilva, 2023-06-01:
I'm trying to use the python-can listener  MF4 (Measurement Data Format v4).

I can use it with no problems without the option "database" in the constructor.

The option "database", in my case, is dedicated to decode J1939 messages with a standard DBC file.

My problem is: 

When I pass the parameter "database" pointing to my DBC file, the error in the title of this issue arise. Simple like that.

ERROR: AttributeError: 'Channel' object has no attribute 'attachments'

Anyone could give me some help ?

CODE THAT IS WORKING:

`
class CAN:
    def __init__(self):
        pass

    async def can_collector(self) -> None:
        can_bus = can.interface.Bus(bustype="socketcan", channel="can1", bitrate=500000, can_filters=can_filter)          
        
        mf4_builder = can.MF4Writer(
            file="mf4/can_collector.mf4",            
            mode="wb",
        )
        
        listeners: List[MessageRecipient] = [
            mf4_builder,  # Regular Listener object
        ]
        
        notifier = can.Notifier(can_bus, listeners)  
        
        print("NOTIFIER CREATED SUCESSFULLY")

if __name__ == "__main__":
    client_can = CAN()

    asyncio.run(client_can.can_collector())
`

CODE THAT IS NOT WORKING:

`
class CAN:
    def __init__(self):
        pass

    async def can_collector(self) -> None:      

        can_bus = can.interface.Bus(bustype="socketcan", channel="can1", bitrate=500000, can_filters=can_filter)  

        mf4_builder = can.MF4Writer(
            file="mf4/can_collector.mf4",
            database="dbc/J1939.dbc",
            mode="wb",
        )

        listeners: List[MessageRecipient] = [
            mf4_builder,  # Regular Listener object
        ]

        notifier = can.Notifier(can_bus, listeners)  
       
        print("NOTIFIER CREATED SUCESSFULLY")
  

if __name__ == "__main__":
    client_can = CAN()

    asyncio.run(client_can.can_collector())

`

I expected the same behavior both cases, but, for some reason, this is not happening.

Some infos regarding my environment:

OS and version: Debian 10
Python version: 3.7
python-can version: 4.2.1

Detailded Traceback and logs:

`
arxml is not supported
ldf is not supported
kcd is not supported
fibex is not supported
xls is not supported
xlsx is not supported
yaml is not supported
odx is not supported
/mnt/SD/bvdc/dbc/J1939.dbc
Traceback (most recent call last):
  File "can_mf4.py", line 118, in <module>
    asyncio.run(client_can.can_collector())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "can_mf4.py", line 91, in can_collector
    mode="wb",
  File "/mnt/SD/venv/lib/python3.7/site-packages/can/io/mf4.py", line 149, in __init__
    source=acquisition_source,
  File "/mnt/SD/venv/lib/python3.7/site-packages/asammdf/blocks/mdf_v4.py", line 2952, in append
    inval_cntr,
  File "/mnt/SD/venv/lib/python3.7/site-packages/asammdf/blocks/mdf_v4.py", line 4562, in _append_structure_composition
    ch = Channel(**kwargs)
  File "/mnt/SD/venv/lib/python3.7/site-packages/asammdf/blocks/v4_blocks.py", line 931, in __init__
    self.attachments = 1
AttributeError: 'Channel' object has no attribute 'attachments'
SocketcanBus was not properly shut down
`

I would appreciate some help in this case.

Thank you,

Henrique

zariiii9003, 2023-06-01:
Hi, what's your asammdf version?

henrique-rsilva, 2023-06-01:
Hello @zariiii9003 , see below:

asammdf==6.2.0

Tks

zariiii9003, 2023-06-01:
Can you confirm the behavior with asammdf 7.3.12?

zariiii9003, 2023-06-04:
Thank you for the feedback, i will close this then.

closed 2023-06-04 by zariiii9003