hardbyte/python-can #1614: find_available_interfaces() doesn't work if virbr0 device exists

nrathaus, 2023-06-13:
### Describe the bug
The find_available_interfaces() fails (with an exception) to work if a virtual device exists in the system:
```
 ip link list up
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: wlp59s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1300 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether c0:b6:f9:bc:aa:c5 brd ff:ff:ff:ff:ff:ff
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 02:42:ae:59:b0:72 brd ff:ff:ff:ff:ff:ff
30: veth2d46635@if29: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP mode DEFAULT group default 
    link/ether 7a:39:f7:d8:53:bd brd ff:ff:ff:ff:ff:ff link-netnsid 0
```

and in JSON format:
```
[{"ifindex":1,"ifname":"lo","flags":["LOOPBACK","UP","LOWER_UP"],"mtu":65536,"qdisc":"noqueue","operstate":"UNKNOWN","linkmode":"DEFAULT","group":"default","txqlen":1000,"link_type":"loopback","address":"00:00:00:00:00:00","broadcast":"00:00:00:00:00:00"},{"ifindex":2,"ifname":"wlp59s0","flags":["BROADCAST","MULTICAST","UP","LOWER_UP"],"mtu":1300,"qdisc":"noqueue","operstate":"UP","linkmode":"DORMANT","group":"default","txqlen":1000,"link_type":"ether","address":"c0:b6:f9:bc:aa:c5","broadcast":"ff:ff:ff:ff:ff:ff"},{},{"ifindex":4,"ifname":"docker0","flags":["BROADCAST","MULTICAST","UP","LOWER_UP"],"mtu":1500,"qdisc":"noqueue","operstate":"UP","linkmode":"DEFAULT","group":"default","link_type":"ether","address":"02:42:ae:59:b0:72","broadcast":"ff:ff:ff:ff:ff:ff"},{"ifindex":30,"link_index":29,"ifname":"veth2d46635","flags":["BROADCAST","MULTICAST","UP","LOWER_UP"],"mtu":1500,"qdisc":"noqueue","master":"docker0","operstate":"UP","linkmode":"DEFAULT","group":"default","link_type":"ether","address":"7a:39:f7:d8:53:bd","broadcast":"ff:ff:ff:ff:ff:ff","link_netnsid":0}]
```

You might notice that there are 'empty' items `{}` in the returned output

### To Reproduce
Easiest way to get a virbr device is to have 'docker' installed on your machine.

### Expected behavior
A list of devices is returned.

### Additional context

Python version: 3.11.2
python-can version: 4.2.0
python-can interface/s (if applicable):

### Suggested fix
```
def find_available_interfaces() -> List[str]:
    """Returns the names of all open can/vcan interfaces

    The function calls the ``ip link list`` command. If the lookup fails, an error
    is logged to the console and an empty list is returned.

    :return: The list of available and active CAN interfaces or an empty list of the command failed
    """

    try:
        command = ["ip", "-json", "link", "list", "up"]
        output_str = subprocess.check_output(command, text=True)
    except Exception:  # pylint: disable=broad-except
        # subprocess.CalledProcessError is too specific
        log.exception("failed to fetch opened can devices from ip link")
        return []

    try:
        output_json = json.loads(output_str)
    except json.JSONDecodeError:
        log.exception("Failed to parse ip link JSON output: %s", output_str)
        return []

    log.debug(
        "find_available_interfaces(): detected these interfaces (before filtering): %s",
        output_json,
    )

    new_json = []
    for item in output_json:
        if 'ifname' not in item:
            continue
        new_json.append(item)

    output_json = new_json

    interfaces = [i["ifname"] for i in output_json if i["link_type"] == "can"]
    return interfaces
```

nrathaus, 2023-06-13:
I wrote a dirty patch, probably a 'smaller' one can be done

Or maybe at the `[...]` section at the bottom you can check if 'ifname' exists

zariiii9003, 2023-06-13:
Maybe we should create backport the fix and release 4.2.2 if this is a frequent problem 🤔

nrathaus, 2023-06-13:
Hi,

Happy to see that someone else has stumbled up on, hopefully this will soon
get addressed in the official release

zariiii9003, 2023-06-18:
@nrathaus try 4.2.2

nrathaus, 2023-06-25:
It worked. Thanks.

closed 2023-06-25 by zariiii9003