hardbyte/python-can #1617: With NeoVI interface, CAN FD payload is blank

dwood-appareo, 2023-06-22:
### Describe the bug
Using python-can on raspberry pi and valueCAN 4 hardware, with neovi bustype.

CAN FD frames have an empty payload.

### To Reproduce
I initialize the bus as follows:

bus = can.interface.Bus(channel=1, bitrate=500000, data_bitrate=2000000, bustype='neovi', fd = True)

Construct a message as follows:

tx_msg = Message(arbitration_id=0x18FFFFFD, data=[0xFE]*64, channel =1, is_fd=True)

Then on the target side I receive blank payloads:

18FFFFFD   [0] 
18FFFFFD   [0]

Sanity check with CAN 2.0:

bus = can.interface.Bus(channel=1, bitrate=500000,  bustype='neovi')

Construct a message as follows:

tx_msg = Message(arbitration_id=0x18FFFFFE, data=[0xFE]*8, channel =1)

Then on the target side I receive the correct frames:

18FFFFFE   [8]  FE FE FE FE FE FE FE FE 
18FFFFFE   [8]  FE FE FE FE FE FE FE FE

### Expected behavior
The expected behavior is that the received frame would be the same length as the transmit frame:

18FFFFFE   [64]  FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE FE

### Additional context

OS and version: Raspbian GNU/Linux 10 (buster)
Python version: 3.7.3
python-can version: python-can==4.0.0
python-can interface/s (if applicable):
valueCAN 4-2 FD

zariiii9003, 2023-06-22:
ping @pierreluctg

pierreluctg, 2023-06-22:
@dwood-appareo are you able to read the messages from the python-can side to confirm if the message read from the bus on the valueCAN adapter side contains data of not?

You can for example simply attach a notifier and listener to you bus in your exiting code and see what the listener is getting.

pierreluctg, 2023-06-22:
@dwood-appareo can you confirm the version of python-can, python-ics and version/name of the driver that you use?

dwood-appareo, 2023-06-22:
Yes I did.

63c81b1c3d13bb96371611d7e2573d6e4adcb6e2

pierreluctg, 2023-06-22:
ok, since your code works on Windows, I do not think that the issue is in the python-can layer.
I can try to reproduce your issue with the same libicsneo, but it may have to wait a bit.

pierreluctg, 2023-06-26:
Reference to python-ics issues raised by @dwood-appareo
https://github.com/intrepidcs/python_ics/issues/150

pierreluctg, 2023-06-29:
Closing sine the issue was identified and resolved in libicsneo (https://github.com/intrepidcs/libicsneo/commit/9ef93eb73e1ada72ccc7a571744b9db1a564f18b) and bundled in python-ics v912.4post1

closed 2023-06-29 by pierreluctg