hardbyte/python-can #1621: AttributeError: module 'can.interfaces' has no attribute 'serial'

WenTheProgrammer, 2023-06-27:
I am completely new to the CAN space so please excuse my ignorance.

I have a controller/computer with an Ubuntu 22.04.2 LTS OS. My Python version is 3.10.6. I believe my controller does not support SocketCAN. After setting up driver and library for the CAN bus port of my controller, I think the CAN bus port name is /dev/ttyUSB0.

Now I would like to simply send a random message through the CAN bus port in Python, and I thought I should follow the "CAN over Serial" tutorial (https://python-can.readthedocs.io/en/master/interfaces/serial.html#). However, after running `can.interfaces.serial.serial_can.SerialBus('/dev/ttyUSB0', baudrate=500000, timeout=0.1, rtscts=False)`, I got the error 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'can.interfaces' has no attribute 'serial'
```
By the way I am using the USB-CANmodul1 from SYS TEC electronic AG to receive the CAN message on my Windows laptop.

Any help is much appreciated!

zariiii9003, 2023-06-28:
You can do this
```python3
import can.interfaces.serial.serial_can
bus = can.interfaces.serial.serial_can.SerialBus('/dev/ttyUSB0', baudrate=500000, timeout=0.1, rtscts=False)
```

or this
```python3
from can.interfaces.serial import SerialBus
bus = SerialBus('/dev/ttyUSB0', baudrate=500000, timeout=0.1, rtscts=False)
```

or this
```python3
import can
bus = can.Bus(interface="serial", channel="/dev/ttyUSB0", baudrate=500000, timeout=0.1, rtscts=False)
```

WenTheProgrammer, 2023-07-06:
Hi @zariiii9003,

All three methods would give me the following error:

`can.exceptions.CanInterfaceNotImplementedError: the serial module is not installed`

I guess I am also not sure whether I should use the serial method in the first place. I would appreciate if you can provide further guidance.

zariiii9003, 2023-07-07:
You need to install [pyserial](https://pypi.org/project/pyserial/)

WenTheProgrammer, 2023-07-07:
Hi @zariiii9003,

I have installed pyserial and now I am able to run the following code on my Ubuntu controller with no error:
```
import can
can_bus = can.interface.Bus(channel='/dev/ttyUSB0', bustype='serial', bitrate=500000)
message_id = 0x1A0
data = [0x33] * 8
message = can.Message(arbitration_id=message_id, data=data)
can_bus.send(message)
```
However, I am unable to receive the sent message on my Windows computer.

Just in case this might be helpful, I am able to use the sample code below from the controller company to send message and receive the message successfully on my Windows computer:
```
#include "wdt_dio.h"
#include <unistd.h>
#include <stdio.h>

int main(int argc, char **argv)
{
	while(1)
	{
		printf("Running\n");
		CAN_SETUP setup;

		setup.bitRate = 500000;//1 Mbps
		printf("channel = %i; _bitRate = %i\n", 0, setup.bitRate);
		if (setup.bitRate == 0)
		{
			printf("ERROR: Unsupported bit rate\n");
			return -1;
		}
		
		setup.recvConfig = CAN_MSG_USE_ID_FILTER;
		setup.recvId = 0;
		setup.recvMask = 0;

		if (!CAN_Setup(0, &setup, sizeof(setup)))
		{
			printf("ERROR: CAN_Setup() failed\n");
			return -1;
		}

		if (!CAN_Start(0))
		{
			printf("ERROR: CAN_Start() failed\n");
			return -1;
		}
		
CAN_MSG  txMsg;
//memset(&txMsg, 0, sizeof(txMsg));
txMsg.id = 0x1A1;
//txMsg.flags = CAN_MSG_EXTENDED_ID;
txMsg.len = 8;
for (int i=0; txMsg.len > i; ++i )
{
	txMsg.data[i] = 0x34;
}

if ( ! CAN_Send(0, &txMsg, sizeof(txMsg)) )
{
	printf("CAN_MessageSet --> FAILED\n");
	return -1;
}	
		sleep(0.5);
		CAN_Stop(0);
		printf("Success!!\n");
	}
	return 0;
}
```

Aziz345987537, 2023-07-11:
iam getting the error no attribute found AttributeError: module 'serial' has no attribute 'serial'
and I cant create object >>ser = serial.serial_for_url(port="COM1", baudrate=9600, bytesize=8, timeout=2,
                            stopbits =serial.STOPBITS_ONE)
i have installed pyserial too 
and i am currently using pycharm
could anyone please help me out

Aziz345987537, 2023-07-11:
and suggest me regarding creating a GUI 
1. Find number of serial ports available and select com port
2. Configure serial port with defined buad rate settings (baud, start bit, stop bit etc)
3. Then write write to serial port and read response from serial port

arbarad, 2023-10-23:
hi @Aziz345987537 

you can use Tkinter library to create GUI using python.

arbarad, 2023-10-23:
Hi @zariiii9003 @WenTheProgrammer @Aziz345987537 

I would like to simply send a random message through the CAN bus port in Python and i used CAN over Serial.

However, after running this code :

import can
import time
import os
with can.Bus() as bus:
    msg = can.Message(
    arbitration_id=0x25,
    data=[0, 25, 0, 1, 3, 1, 4, 1],
    is_extended_id=True
    )
    try:
        bus.send(msg)
        print(f"Message sent on {bus.channel_info}")
    except can.CanError:
        print("Message NOT sent")

I got :
Traceback (most recent call last):
  File "C:\Users\arbarad\Python311\Python311\python\programes\serial_port.py", line 162, in <module>
    with can.Bus() as bus:
         ^^^^^^^
AttributeError: module 'can' has no attribute 'Bus'

and i am currently using visual studio
could anyone please help me out

closed 2023-12-29 by WennPaper