hardbyte/python-can #1622: when using qtdesigner transferring to exe file(executative file), the hardware cannot be detected?

leijr, 2023-06-28:
### Describe the bug

when i develop it in pycharm, everything runs OK，but when i generate the exe file, hardware cannot de detected.

### To Reproduce

### Expected behavior

### Additional context

OS and version: win 10
Python version: 3.7
python-can version: 4.2.0
python-can interface/s (if applicable): vector/pcan

<details><summary>Traceback and logs</summary>

        config_contents = 'vector-channel-1'
        config_contents_list = [i for i in config_contents.split('-')]
        name_config = config_contents_list[0]
        channel_config = config_contents_list[-1]
        if self.can_type == 'canfd':
            try:
                self.is_fd = True
                self.dlc = 64
                if name_config == 'vector':
                    channel_num = int(channel_config) - 1
                    self.bus = can.interface.Bus(channel=channel_num, interface="vector", can_filters=self.filters,
                                                 set_receive_own_message=True,
                                                 fd=True, bitrate=500000, data_bitrate=2000000, sjw_abr=32,
                                                 tseg1_abr=127,
                                                 tseg2_abr=32, sam_abr=1, sjw_dbr=12, tseg1_dbr=27, tseg2_dbr=12)
                    return True
                elif name_config == 'pcan':
                    self.bus = can.interface.Bus(
                        bustype='pcan', channel=channel_config,
                        set_receive_own_message=True, fd=True, f_clock_mhz=80,
                        nom_brp=1, nom_tseg1=127, nom_tseg2=32, nom_sjw=32,
                        data_brp=1, data_tseg1=27, data_tseg2=12, data_sjw=12)
                    return True

</details>

luojiaaoo, 2023-09-26:
It is normal for me to detect vector with pyinstaller package. What do you use package tool?

leijr, 2023-09-30:
> It is normal for me to detect vector with pyinstaller package. What do you use package tool?
the same, pyinstaller.  do you have any special steps when packaging using pyinstaller?
i just follow the instructions from the internet:
 1. generating the spec file
 2. change the spec file log( insert the dll file information into the line,  datas=[.........])
 3. compile the spec file again
  tried thousands of times,but it didn't work..
would you tell me what did you do on how to package file?  or the refrence network link
and what kind of the hard config method did you apply in you script, i copied the python-can officail instructions on the below. 
 i applied the second, -> bus = can.interface.Bus(interface='socketcan', channel='vcan0', bitrate=500000) <-
i would really appreciate if you can give some tips!!!!!!!!!!!!!!! thanks you very much
{
2.1 In Code
The can object exposes an rc dictionary which can be used to set the interface and channel.
**import can
can.rc['interface'] = 'socketcan'
can.rc['channel'] = 'vcan0'
can.rc['bitrate'] = 500000
from can.interface import Bus
bus = Bus()**
You can also specify the interface and channel for each Bus instance:
**import can
bus = can.interface.Bus(interface='socketcan', channel='vcan0', bitrate=500000)**
2.2 Configuration File
On Linux systems the config file is searched in the following paths:
**1. ~/can.conf
2. /etc/can.conf
3. $HOME/.can
4. $HOME/.canrc**
On Windows systems the config file is searched in the following paths:
1. %USERPROFILE%/can.conf
2. can.ini (current working directory)
3. %APPDATA%/can.ini
}

FedericoSpada, 2024-04-04:
Hi all,
I've encountered the same problem: at startup, my GUI scans for all connected dongles using `detect_available_configs()` so that the user can choose them freely. By running the files via the command line I can detect both Ixxat and Vector dongles. If I create the .exe using pyinstaller and his .spec file, only the Ixxat dongles can be detected in the standalone app.
I've debugged the `can` module, and the problem is within the `_get_class_for_interface` function when the Vector module is imported thanks to `importlib.import_module`: it results that it is not found within the .exe file.

I suspect that pyinstaller is not able to detect this weak import, so for it, those files are not used and so not added inside the .exe file. For the Ixxat dongle, within ixxat\canlib.py there is an explicit import statement that I think is the reason for which pyinstaller add those files to the exe; instead, in vector\canlib.py, there is no explicit import.
If I write `import can.interfaces.vector` somewhere in my code, pyinstaller is forced to add that module inside the .exe and everything works fine. The problem is that I have to do it for all interfaces. I've tried to do it inside the `can` module but I've encountered a recursive import problem that I'm not sure how to fix.

For now, I've solved the problem by compiling the `hiddenimports` field within the .spec file, so that to force pyinstaller to add all interfaces' modules.

leijr, 2024-04-05:
Thank you very much!
Get it done!!

closed 2024-04-05 by leijr