hardbyte/python-can #1624: tests fail on bigendian

umlaeute, 2023-06-29:
### Describe the bug

Running the test-suite on a BingEndian architecture (like `s390x`), fails in two places:
- test_bit_timing.py::test_btr_persistence FAILED                                                                                                                                                                                      
- test_pcan.py::TestPCANBus::test_detect_available_configs FAILED

```
90s =================================== FAILURES ===================================
 90s _____________________________ test_btr_persistence _____________________________
 90s 
 90s     def test_btr_persistence():
 90s         f_clock = 8_000_000
 90s         for btr0btr1 in PCAN_BITRATES.values():
 90s             btr1, btr0 = struct.unpack("BB", btr0btr1)
 90s     
 90s >           t = can.BitTiming.from_registers(f_clock, btr0, btr1)
 90s 
 90s test_bit_timing.py:180: 
 90s _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
 90s /usr/lib/python3/dist-packages/can/bit_timing.py:183: in from_registers
 90s     return cls(
 90s /usr/lib/python3/dist-packages/can/bit_timing.py:70: in __init__
 90s     self._validate()
 90s _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
 90s 
 90s self = can.BitTiming(f_clock=8000000, brp=21, tseg1=1, tseg2=1, sjw=1, nof_samples=1)
 90s 
 90s     def _validate(self) -> None:
 90s         if not 8 <= self.nbt <= 25:
 90s >           raise ValueError(f"nominal bit time (={self.nbt}) must be in [8...25].")
 90s E           ValueError: nominal bit time (=3) must be in [8...25].
 90s 
 90s /usr/lib/python3/dist-packages/can/bit_timing.py:74: ValueError
 90s __________________ TestPCANBus.test_detect_available_configs ___________________
 90s 
 90s self = <test.test_pcan.TestPCANBus testMethod=test_detect_available_configs>
 90s 
 90s     def test_detect_available_configs(self) -> None:
 90s         if platform.system() == "Darwin":
 90s             self.mock_pcan.GetValue = Mock(
 90s                 return_value=(PCAN_ERROR_OK, PCAN_CHANNEL_AVAILABLE)
 90s             )
 90s             configs = PcanBus._detect_available_configs()
 90s             self.assertEqual(len(configs), 50)
 90s         else:
 90s             value = (TPCANChannelInformation * 1).from_buffer_copy(
 90s                 b"Q\x00\x05\x00\x01\x00\x00\x00PCAN-USB FD\x00\x00\x00\x00"
 90s                 b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
 90s                 b'\x00\x00\x00\x00\x00\x00\x003"\x11\x00\x01\x00\x00\x00'
 90s             )
 90s             self.mock_pcan.GetValue = Mock(return_value=(PCAN_ERROR_OK, value))
 90s >           configs = PcanBus._detect_available_configs()
 90s 
 90s test_pcan.py:382: 
 90s _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
 90s 
 90s     @staticmethod
 90s     def _detect_available_configs():
 90s         channels = []
 90s         try:
 90s             library_handle = PCANBasic()
 90s         except OSError:
 90s             return channels
 90s     
 90s         interfaces = []
 90s     
 90s         if platform.system() != "Darwin":
 90s             res, value = library_handle.GetValue(PCAN_NONEBUS, PCAN_ATTACHED_CHANNELS)
 90s             if res != PCAN_ERROR_OK:
 90s                 return interfaces
 90s             channel_information: List[TPCANChannelInformation] = list(value)
 90s             for channel in channel_information:
 90s                 # find channel name in PCAN_CHANNEL_NAMES by value
 90s >               channel_name = next(
 90s                     _channel_name
 90s                     for _channel_name, channel_id in PCAN_CHANNEL_NAMES.items()
 90s                     if channel_id.value == channel.channel_handle
 90s                 )
 90s E               StopIteration
 90s 
 90s /usr/lib/python3/dist-packages/can/interfaces/pcan/pcan.py:691: StopIteration
 90s =========================== short test summary info ============================
 90s FAILED test_bit_timing.py::test_btr_persistence - ValueError: nominal bit tim...
 90s FAILED test_pcan.py::TestPCANBus::test_detect_available_configs - StopIteration
 90s ================== 2 failed, 427 passed, 97 skipped in 32.84s ==================
```

### To Reproduce

Run the test-suite on a BigEndian machine

### Expected behavior

tests should succeed, regardless of endianness

### Additional context

| component | version |
|---|---|
|OS and version| Debian/trixie |
|Python version| 3.11 |
|python-can version| 4.2.2 |

umlaeute, 2023-06-29:
## test_btr_persistence

https://github.com/hardbyte/python-can/blob/dc0ae68acb28f01a503e084718cfa2acf39d1e16/test/test_bit_timing.py#L177-L178

on BigEndian, this really returns `btr0, btr1`.

Is the assumption of the byte-ordering broken in the test (as is indicated by the naming of the variables), or should `PCAN_BITRATE` hold normalized values in the first place?

zariiii9003, 2023-06-29:
That part can probably be solved with a `<`:
```python3
 for btr0btr1 in PCAN_BITRATES.values(): 
     btr1, btr0 = struct.unpack("<BB", btr0btr1) 
```

But the `can.BitTiming.from_registers` method probably needs to be adapted too. Using `ctypes.LittleEndianStructure` instead of bit masks and bit shifts might help.

umlaeute, 2023-06-29:
now really (in my tests).
it can be solved with
```
btr0, btr1 = struct.pack(">H", btr0btr1.value)
```

umlaeute, 2023-06-29:
sure.

any ideas about the *other* issue?
i guess that https://github.com/hardbyte/python-can/blob/dc0ae68acb28f01a503e084718cfa2acf39d1e16/test/test_pcan.py#L381-L385 encodes the data in little endian as well...

umlaeute, 2023-06-29:
generating the data with the correct endianness solves the second issue:

```python
data=struct.pack("HBBI33sII", 81, 5, 0, 1, b'PCAN-USB FD', 1122867, 1)
value = (TPCANChannelInformation * 1).from_buffer_copy(data)
```

@zariiii9003 does this look good to you?

closed 2023-06-30 by zariiii9003