hardbyte/python-can #1630: wrong justing of ID column in __str__ method of Message class

michalPradzynski, 2023-07-05:
### Describe the bug
We are using __ str __ method of Message class in the logging of our project and found out that when there are frames with short and extended Arbitration IDs, short ID ("ID: dddd") is justed to the right of its' column which makes the ID column distorted.
The "ID: " part for both short and extended AIDs is not in the same vertical line.

### To Reproduce

- create a logger which logs each CAN frame with Arbitration IDs with different lengths as str(can.Message)
- create a virtual CAN bus
- send some CAN frames

### Expected behavior
Change arbitration_id_string.rjust() to arbitration_id_string.ljust in line 116 of can/message.py
https://github.com/hardbyte/python-can/blob/c5d7a7ce71e5e12cad4682d6c1411f8688de54bd/can/message.py#L110-L116

### Additional context

OS and version: Windows 10 Enterprise 21H2
Python version: 3.10
python-can version: 4.1.0

<details><summary>Traceback and logs</summary>

How the logs look right now:
[image]

zariiii9003, 2023-07-07:
Would you like to create a PR?

closed 2023-07-11 by zariiii9003

michalPradzynski, 2023-07-13:
> Would you like to create a PR?

Well, you beat me up to it, so the only think I can do now is thank you :)

PR #1635 (2023-07-11): Align `ID:` in message string
Close #1630