hardbyte/python-can #1637: Does it support feature somelike Tx-Self Ack?

xRowe, 2023-07-18:
Dear
   
    I write a script to read DTC using the awesome project. 
    But my script may send requests while the ECU are all sleep and CANazlyzer project is running, At this time, when the request is sent, there will be no acknowledge received, and Vector CANcase/CANazlyer will be eventually stuck(ChipState become passive, no message can be sent out then) until restart the project or `ResetCan` request by CAPL.
    So I wonder if there is Tx-Self Ack feature like in this project avoid this situation? 
    Sorry, I read the doc but I got nothing, so I come here?

Thank you very much

leijr, 2023-08-25:
try to set the following parameter as true:
receive_own_message = True or something like this
check the exact grammer usage on official document .

hartkopp, 2023-08-25:
"receive_own_message = True" makes the system to echo the sent CAN frame to the (sending) application which is unusual and not solving the described problem.

@xRowe : Tx-Self ACK is a feature of (only some!) CAN controllers to run without other nodes on the CAN bus. But it is not defined whether there is some sending on the bus.

When you send a CAN frame the ECU normally instantly wakes up to communicate with you.  So there should be no  problem. If this is not guaranteed I would suggest to add a second CAN controller to the CAN bus (AFAIK Vector CAN adapters always come with two channels). Then you always have a properly working CAN bus - no matter if your ECU starts in time.

luojiaaoo, 2023-09-13:
Maybe you could set "timeout" for send, and execpt Exception to information about success or not

xRowe, 2023-10-20:
> "receive_own_message = True" makes the system to echo the sent CAN frame to the (sending) application which is unusual and not solving the described problem.
>
> @xRowe : Tx-Self ACK is a feature of [...]

When Using CANalyzer or CANoe, there will be an option call "Tx-Self Ack".
If it is checked, the Tx will ignore ack
So I wanna know if there is a tricky way can acheive this, by Vector interface