hardbyte/python-can #1639: Validate arbitration ID.

joooeey, 2023-08-02:
### Is your feature request related to a problem? Please describe.
When I try to send a message with an arbitration IDENTIFIER that is not allowed by the CAN standard, my hardware* doesn't react (silent failure). The arbitration identifier should be `0` to `0x07ff` for the standard frame format and `0` to `0x1fffffff` for the extended format. Source: [CAN Specification 2.0](https://web.archive.org/web/20221010170747/http://esd.cs.ucr.edu/webres/can20.pdf), Sections 3.1.1 and 3.2.1.

### Describe the solution you'd like
Check the arbitration identifier when creating a message (depending on supplied frame format). And raise a `ValueError` if the input value for `arbitration_id` is out of range. 

### Describe alternatives you've considered

- Leave it up to the user to pass in valid values. That adds an unnecessary potential point of failure.
- Implement the value checking for each interface. That adds an enormous amount of code duplication.

### Additional context
- (*) Hardware: SUNIX CAN card 5C50W32055
- interface: self-coded
- Hardware and interface should be irrelevant to the feature request itself because the valid values are already defined in the original BOSCH standard. I assume that ISO 11898-1 doesn't differ much but I can't afford that document.

zariiii9003, 2023-08-02:
did you try setting `can.Message(..., check=True)`? See [docs](https://python-can.readthedocs.io/en/stable/message.html#can.Message).

joooeey, 2023-08-03:
> did you try setting `can.Message(..., check=True)`? See [docs](https://python-can.readthedocs.io/en/stable/message.html#can.Message).

No. That does indeed solve the issue. I was however surprised that the error is only raised when calling send. Is there a particular reason this isn't thrown when the `can.Message` is created (fail early paradigm)?

zariiii9003, 2023-10-08:
> No. That does indeed solve the issue. I was however surprised that the error is only raised when calling send. Is there a particular reason this isn't thrown when the `can.Message` is created (fail early paradigm)?

Mainly for performance reasons, there are use cases where many thousands of `Message` instances are created each second. The validation would be a waste of resources.

closed 2023-10-08 by zariiii9003