hardbyte/python-can #1642: Invalid version error when initializing the CAN bus using can.interface.Bus

lcc2428, 2023-08-14:
### Describe the bug

Hello,

I have been using the python-can package with the PEAK interface for a couple months now for a project that I have been working on (have been using the PyCharm IDE). Up until this point, I have not had any issues. However, whenever I now try to initialize the CAN bus using can.interface.Bus, I get an invalid version error. I have spent the past few days researching the issue and what might be causing it, but haven't had any luck. I am not entirely sure if this is directly an issue with the python-can package or if this just might be that something updated in my environment and now isn't compatible with the package; regardless, I would greatly appreciate any guidance that you could provide me with. Below I have pasted the simple program that I have been using for trying to trace the issue, along with the related error message that arises when I run the code/program.

### Expected behavior
I have been using the python-can package for the last two months on this project and did not have any issues until around midday on Friday. Ran my program before a meeting ~10am and everything worked as I expected. I came back from the meeting around an hour after to find that my program no longer worked (I didn't touch any of my code, so was very confused on how it would just randomly not be working anymore). Found that the issue is directly related to the initialization of the CAN bus and that there is some version mismatch (or possibly a parsing issue) that is causing the error?

### Additional context

OS and version: Windows 10
Python version: 3.11
python-can version: 4.2.2
setuptools: 65.5.0
packaging: 23.1
python-can interface/s: PEAK

<details><summary>Traceback and logs</summary>

 #### Current basic program called versionTests.py that is being used for testing/tracing:

import can

tpd = can.interface.Bus(bustype='pcan', channel="PCAN_USBBUS1", bitrate=500000, timestamp=True)
tpd.shutdown()

#### Error message that is produced by running the versionTests.py program:

Traceback (most recent call last):
File "versionTests.py", line 9, in <module>
    tpd = can.interface.Bus(bustype='pcan', channel="PCAN_USBBUS1", bitrate=500000, timestamp=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\can\util.py", line 369, in wrapper
    return f(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\can\interface.py", line 138, in __new__
    bus = cls(channel, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\can\interfaces\pcan\pcan.py", line 259, in __init__
    self.check_api_version()
  File "venv\Lib\site-packages\can\interfaces\pcan\pcan.py", line 414, in check_api_version
    apv = self.get_api_version()
          ^^^^^^^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\can\interfaces\pcan\pcan.py", line 411, in get_api_version
    return version.parse(value.decode("ascii"))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\packaging\version.py", line 52, in parse
    return Version(version)
           ^^^^^^^^^^^^^^^^
  File "venv\Lib\site-packages\packaging\version.py", line 198, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
packaging.version.InvalidVersion: Invalid version: '1, 3, 0, 50'
PcanBus was not properly shut down

</details>

zariiii9003, 2023-08-14:
Thanks for reporting. Your value does not comply with the PCAN docs: `The API version value is represented as a string of the form “a.b.c.d”`, but that should be an easy fix.

zariiii9003, 2023-08-14:
I think your driver version is to old. A minimum version of 4.2 was suggested in #1188

lcc2428, 2023-08-14:
> I think your driver version is too old. A minimum version of 4.2 was suggested in #1188

My PEAK Driver is version 4.4.0.

lcc2428, 2023-08-14:
Just solved the issue, you were absolutely right. My version was up to date but it needed repaired. Thank you!

closed 2023-08-14 by lcc2428