hardbyte/python-can #1646: Replay candump: `MessageSync` (python-can) and   `canplayer` (can-utils) behave slightly differently

yomannnn, 2023-08-17:
Hi,
### Observation:  
Regarding the skip parameter  `MessageSync` (python-can) and   `canplayer` (can-utils) behave slightly differently.

`canplayer -s 1 -I candump.log`
skip gaps in timestamps > 1 second
if skip gap  is detected  the subsequent CAN msg is sent *instantly*

however 

`MessageSync(reader,timestamps=True, skip=1)`
skip gaps in timestamps > 1 second
if skip gap  is detected  the subsequent CAN msg is sent only *after 1sec*

As a result there is an additional  gap of 1sec if I compare both traces.

### Expected behavior
`canplayer` and `python-can`  replay stored canlogs  exactly the same

### Additional context
I'm not really sure whether this is a bug or a feature.
However I decided to share my experience as I spent quite some time to find the root cause of my problem.
Many thanks ...

zariiii9003, 2023-08-18:
I don't really consider this a bug, since we do not actually promise to imitate the behavior of canplayer. That said, you're welcome to provide a pull request to change this.

hartkopp, 2023-08-18:
I also had to read the issue once again.
What is the intention for `MessageSync(reader,timestamps=True, skip=1)` then ?

If it should be really "skip gaps in timestamps > 1 second" these gaps should be squashed to zero which would lead to an immediate processing of the frame following the gap.

So what would be the benefit to wait for another second before processing the frame after the gap then?