hardbyte/python-can #1648: PEAK driver not found in 32bits environment

Svalann, 2023-08-22:
### Describe the bug

I can't connect with my PEAK-USB dongle when in my 32 bit environment, there is no issue when using 64 bit environment.
I get OSError: The PEAK-driver could not be found!

There is no issues with Kvaser, Vector, IXXAT or Iscan.

### To Reproduce

Python-can is used through canopen for python ([https://github.com/christiansandberg/canopen](url))
Using 32 bits python just trying to do a connect.

### Additional context
The issue seems to be that the check to look for the driver fails even though it exists. If check is ignored it successfully loads the driver. It fails on "with winreg.OpenKey(reg, r"SOFTWARE\PEAK-System\PEAK-Drivers"):" If raise OSError is replaced with "pass" it will successfully load the driver.

```python
        if platform.system() == "Windows":
            load_library_func = windll.LoadLibrary

            # look for Peak drivers in Windows registry
            with winreg.ConnectRegistry(None, winreg.HKEY_LOCAL_MACHINE) as reg:
                try:
                    with winreg.OpenKey(reg, r"SOFTWARE\PEAK-System\PEAK-Drivers"):
                        pass
                except OSError:
                    raise OSError("The PEAK-driver could not be found!") from None
        else:
            load_library_func = cdll.LoadLibrary
```

OS and version: Windows 11 Pro Version:22H2 OS build: 22621.1992
Python version: 3.7.9-32
python-can version: 4.2.2
python-can interface/s (if applicable): PEAK-Drivers 4.4.0.556

<details><summary>Traceback and logs</summary>

  File "C:\MyTool \interface\canopen\network.py", line 109, in connect
    self.bus = can.interface.Bus(*args, **kwargs)
  File "c:\MyTool\python\lib\site-packages\can\util.py", line 369, in wrapper
    return f(*args, **kwargs)
  File "c:\MyTool\python\lib\site-packages\can\interface.py", line 138, in __new__
    bus = cls(channel, **kwargs)
  File "c:\MyTool\python\lib\site-packages\can\interfaces\pcan\pcan.py", line 238, in __init__
    self.m_objPCANBasic = PCANBasic()
  File "c:\MyTool\python\lib\site-packages\can\interfaces\pcan\basic.py", line 688, in __init__
    raise OSError("The PEAK-driver could not be found!") from None
OSError: The PEAK-driver could not be found!

</details>

zariiii9003, 2023-08-23:
That check was added in #724. I wonder if it is really necessary.
But i don't understand why it works in 64bit, but not in 32bit? Did you test on the same machine?

Svalann, 2023-08-24:
I can't figure it out either but I think 64 and 32 store
registry differently and this might affect this. But yes I'm using the same
machine with the environment identical except for running python in either
32 or 64.

https://stackoverflow.com/questions/869783/windows-64-bit-registry-v-s-32-bit-registry

I've seen some examples with similar code so it's probably normal but my
first thought is that *ConnectRegistry *as *reg *seems suspicious.

[image: image.png]

In this screenshot I changed the raise OSError to logger.error. I can't see
why this test is here, you only do a pass and later an OSError is raised if
the library fails to load.

zariiii9003, 2023-08-24:
Does `r"SOFTWARE\Wow6432Node\PEAK-System\PEAK-Drivers"` work in 32bit?

Svalann, 2023-08-24:
No, it does not. But now when I take a look at the registry I can see that the "PEAK-Drivers" folder is missing in the Wow6432Node, but it is present in the "SOFTWARE\PEAK-System\PEAK-Drivers".

I had checked that PEAK was supported with 32 bits but I think I might have been fooled by its description now when I check again. It seems it only supports Windows x64. But it also mention drivers for (32/64-bit).
[PEAK drivers](https://www.peak-system.com/Drivers.523.0.html?&L=1)
From PEAK documentation:

> PCAN-Basic: Installs the PCAN-Basic application programming
> interface (API library), for interfacing with all device drivers of
> PEAK-System. This library is available for x86, x64, and ARM64
> applications.

So I guess the issue is the PEAK installment. I clearly have the PCANBasic.dll for x86 but PEAK is not in the register for x86. Thus the test fails even tough it can find the dll.

Svalann, 2023-09-01:
I had a post at peak discussing this: [PEAK forum](https://forum.peak-system.com/viewtopic.php?f=119&t=7423)

Didn't really get a full answer but they seemed to question why the regkey is used in this case. I myself is not sure if this regkey is even related to the pcanbasic lib but other parts of installing their package since the basic lib supports x32 but not the rest of the package.

zariiii9003, 2023-09-10:
@tamenol  As the author of #724, would you be able to provide a 32bit compatible solution?

PR #1672 (2023-10-08): PCAN: remove Windows registry check
Closes #1648
closed 2023-10-08 by zariiii9003