hardbyte/python-can #1656: vector: ValueError: 16 is not a valid XL_OutputMode

afflux, 2023-09-05:
### Describe the bug
Opening a Vector 1610A CAN FD interface fails with `ValueError: 16 is not a valid XL_OutputMode`

### To Reproduce
Just creating the bus object with `can.Bus(interface="vector", channel="0", fd=True)` is enough to trigger this on my test setup.
Not sure which settings cause that bit to be set in the outputMode field, unfortunately...
### Expected behavior
Creating the bus succeeds.

### Additional context

OS and version: Windows 10
Python version: 3.10
python-can version: 4.2.2
python-can interface/s: Vector VN1610A

Downgrading to 4.0.0 solves the problem (outputMode enum is not parsed there)

<details><summary>Traceback and logs</summary>

```
Traceback (most recent call last)
  File "can\util.py", line 369, in wrapper
  File "can\interface.py", line 138, in __new__
  File "can\util.py", line 369, in wrapper
  File "can\interfaces\vector\canlib.py", line 204, in __init__
  File "can\interfaces\vector\canlib.py", line 1161, in get_channel_configs
  File "can\interfaces\vector\canlib.py", line 1114, in _read_bus_params_from_c_struct
  File "enum.py", line 711, in __call__
  File "enum.py", line 1128, in __new__
ValueError: 16 is not a valid XL_OutputMode
```

</details>

afflux, 2023-09-05:
Yes, seems to be fixed there

j-c-cook, 2023-09-24:
ping @zariiii9003 Please close.

closed 2023-10-08 by zariiii9003