hardbyte/python-can #1661: BLFReader zlib.error: Error -5 while decompressing data: incomplete or truncated stream in Python

ro-id, 2023-09-11:
### Describe the bug
for larger blf files the Error occurs:
zlib.error: Error -5 while decompressing data: incomplete or truncated stream in Python

### To Reproduce
try to read a large blf file

### Expected behavior
failure

### Additional context

OS and version: Windows 10
Python version: 3.10.12
python-can version: 4.2.2
python-can interface/s (if applicable):

compare to issue https://stackoverflow.com/questions/32367005/zlib-error-error-5-while-decompressing-data-incomplete-or-truncated-stream-in

can be solved by replacing:
```python
elif method == ZLIB_DEFLATE:
    data = zlib.decompress(container_data, 15, uncompressed_size)
```
with:
```python
elif method == ZLIB_DEFLATE:
    zobj = zlib.decompressobj()  # obj for decompressing data streams that won’t fit into memory at once.
    data = zobj.decompress(container_data)
```

j-c-cook, 2023-09-24:
ping @zariiii9003 Please close this issue as #1662 has been merged.

j-c-cook, 2023-09-24:
@ro-id See linking keywords https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword

closed 2023-09-24 by ro-id