hardbyte/python-can #1663: Add comments for messages

pkess, 2023-09-14:
### Is your feature request related to a problem? Please describe.
I am using this library to analyze can traces in several formats. Sometimes i would like to add some comments to can messages that are then printed inside an output trace file. This is supported for example by the asc file format.

### Describe the solution you'd like
The class `Message` should get an attribute `comment` by default (e.g. None by default)

### Describe alternatives you've considered
right now i am using some hack to write to the file object inside the writer classes but this is quite unstable when the library changes

### Additional context
I would like to implement this and add it to the reader and writer classes as well but first i would like to get some feedback on this feature request.

j-c-cook, 2023-09-24:
Can you provide an example `.asc` file that contains comments for CAN messages? Which program are you using that supports comments in the `.asc` file format? Are there any other file formats that support comments?

pkess, 2023-09-24:
Hello @j-c-cook unfortunatelly i do not have any asc file that contains comments. But there is a knowledge base article at vector that says it is possible: https://support.vector.com/kb?id=kb_article_view&sysparm_article=KB0011536&sys_kb_id=b6d21110870d4550b9f233770cbb3523&spa=1
And it says the blf format is supporting comments as well

And the trc file format from peak at leas supports comments as separate lines (comment for a message can be added write after the message line for example):
https://www.peak-system.com/produktcd/Pdf/English/PEAK_CAN_TRC_File_Format.pdf