hardbyte/python-can #1674: I can't detect the 582.1 device by can.interface.detect_available_configs(interfaces="etas")

luojiaaoo, 2023-10-10:
### Additional context

OS and version: win10
Python version: 3.7.9 embed
python-can version: 4.2.2

```python
import can
configs = can.interface.detect_available_configs(interfaces="etas")
for c in configs:
    print(c)
```

print:
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:100/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:100/CAN:2'}
{'interface': 'etas', 'channel': 'ETAS://J2534_HostInterface:1/J2534_System:Kvaser AB|J2534 for Kvaser Hardware|0404/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://J2534_HostInterface:1/J2534_System:Shanghai TOSUN Technology Ltd.|TSJ2534|0404/CAN:1'}

INCA can use ES582.1, but is can not be detected.

lumagi, 2023-10-25:
Hi,
I just tested on Windows 10 with an ES582.1, BOA  x64 v2.35.0, python-can 4.2.2 and I get the following output:
```
{'interface': 'etas', 'channel': 'ETAS://USB/ES582.1:00112233/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://USB/ES582.1:00112233/CAN:2'}
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:100/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:100/CAN:2'}
```

It's weird that you do get an output from BOA but it's not listing the ES582.1.
I would have the following questions:
* Do you use different BOA implementations with Busmaster / python-can? I.e. 32bit vs 64bit?
* What version of BOA are you using?

andebjor, 2023-11-03:
We have the same issue: that `python-can` finds _some_ things when searching for `etas` interfaces (virtual Kvaser and Vector devices my case), but not the physical ES582.1 dongle.

When playing around we realized that it is a 32/64 bit issue somehow. With 64-bit Python-3.9 it does _**not**_ work, but with 32-bit Python-3.9 i _**does**_ work. Unfortunately, using a 32-bit Python version is not an option for us for other reasons.

I have BOA v2.35.0 and tested with python-can 4.2.2.

lumagi, 2023-11-04:
> We have the same issue: that `python-can` finds _some_ things when searching for `etas` interfaces (virtual Kvaser and Vector devices my case), but not the physical ES582.1 dongle.
>
> When playing [...]

Sorry to be nitpicky here, but did you install both the newest x86 and x64 version? I don't want to rule out an issue in python-can, but judging from the answer from @luojiaaoo, it might be related to your setup.

andebjor, 2023-11-04:
> did you install both the newest x86 and x64 version?

Yes, I created separate virtualenvs with 32 and 64 bit Python-3.9. I only have the 64-bit BOA installed though (but it includes both 32 and 64 bit libraries it seems). I also tested with 64-bit Python-3.11, with the same result that it can't find the physical device.

I added some printouts to the function that attempts to find the devices, but I couldn't see anything obviously wrong there. But it find nothing on the USB device tree for some reason. Well, in 64-bit. In 32-bit it works as expected.
 
My hunch would be that it's actually an issue inside the (64-bit) BOA libs, but I really don't know.

lumagi, 2023-11-16:
Sorry for replying so late. I really don't know what could be the issue here. I worked fine with x64 on my machine. Can you maybe try a fresh installation? Other than that, I'm kind of clueless.

closed 2024-06-11 by luojiaaoo