hardbyte/python-can #1691: SeeedBus does not transmit bus messages

RyanHope, 2023-11-14:
### Describe the bug
SeeedBus can be used with the logger and viewer scripts but data transmitted via it does not seem to actually reach the canbus. 

### To Reproduce
I have both a Kvaser Leaf and a SeeedStudio adapter connected to the same bus which is connected to an motorcycle ECU. 
If use can.logger to sniff the bus with seedstudio and transmit the commands below, the seedstudio adapters records the messages and the response. If I reverse things and log with kvaser and transmit with seeedstudio, nothing shows up in the logger at all.
```
import can
# with can.Bus(interface='kvaser', channel=0) as bus:
with can.Bus(interface='seeedstudio', channel='COM12') as bus:
    msg = can.Message(arbitration_id=0x18da10f1, data=[0x02, 0x3e, 0x80, 0x55, 0x55, 0x55, 0x55, 0x55], is_extended_id=True, is_rx=False)
    bus.send(msg)
    recv_msg = bus.recv(timeout=1.0)

    msg = can.Message(arbitration_id=0x18da10f1, data=[0x02, 0x3e, 0x00, 0x55, 0x55, 0x55, 0x55, 0x55], is_extended_id=True, is_rx=False)
    bus.send(msg)
    recv_msg = bus.recv(timeout=1.0)
```

### Expected behavior
```
> python -m can.logger -i seeedstudio -c COM12
Connected to SeeedBus: Serial interface: COM12
Can Logger (Started on 2023-11-13 20:58:35.139360)
Timestamp: 1699927130.034850    ID: 18da10f1    X Rx                DL:  8    02 3e 80 55 55 55 55 55
Timestamp: 1699927131.046703    ID: 18da10f1    X Rx                DL:  8    02 3e 00 55 55 55 55 55
Timestamp: 1699927131.051705    ID: 18daf110    X Rx                DL:  8    02 7e 00 aa aa aa aa aa
```

### Additional context

OS and version: Win10
Python version: 3.8 32bit
python-can version: [5c1c46f](https://github.com/hardbyte/python-can/commit/5c1c46fdbd5935c5a7a8b34aa6a8df3af201616b)
python-can interface/s (if applicable): seeedstudio

tsabelmann, 2024-01-01:
For me it looks like that the SeeedStudio interface is not implemented properly - I mean a normal recv function like in the other interface modules is missing. Additionally, the SeeedStudio interface supports two protocols - 20 bytes fixed-size and a variable-length byte-sized protocol. Maybe, if you want, you could implement the missing parts. Feel free to reach out.

vChavezB, 2024-03-06:
I have found that if I add a delay after the init frame

https://github.com/hardbyte/python-can/blob/d40915cae080716293b7293c9261adfe1fdc2936/can/interfaces/seeedstudio/seeedstudio.py#L118

so in practice I initialize the seedstudio can interface and add a delay :

```python
 can.interface.Bus(bustype='seeedstudio',
                                     channel=self.com_port,
                                     baudrate=2000000,
                                     bitrate=500000,
                                     frame_type='STD',
                                     operation_mode='normal')
time.sleep(0.1)                                     
```

I can send can messages. For some reason it seems that if not, the serial write does not send correctly the data.