hardbyte/python-can #1693: In EtasBus _apply_filters, self._oci_filters is not defined when EtasBus init called with version 4.3.0

kjw8118, 2023-11-17:
### Describe the bug

When I generate Bus object with 'etas', it gave me error messages
`dev = can.interface.Bus(interface='etas', channel='ETAS://USB/ES582.1:XXXXXX/CAN:1', bitrate=500000)`

```EtasBus was not properly shut down
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "'Python Path'\lib\site-packages\can\util.py", line 378, in wrapper
    return f(*args, **kwargs)
  File "'Python Path'\lib\site-packages\can\interface.py", line 135, in Bus
    bus = cls(channel, **kwargs)
  File "'Python Path'\lib\site-packages\can\interfaces\etas\__init__.py", line 21, in __init__
    super().__init__(channel=channel, **kwargs)
  File "'Python Path'\lib\site-packages\can\bus.py", line 99, in __init__
    self.set_filters(can_filters)
  File "'Python Path'\lib\site-packages\can\bus.py", line 404, in set_filters
    self._apply_filters(self._filters)
  File "'Python Path'\lib\site-packages\can\interfaces\etas\__init__.py", line 222, in _apply_filters
    if self._oci_filters:
AttributeError: 'EtasBus' object has no attribute '_oci_filters'. Did you mean: 'set_filters'?
```

I used etas interface python-can with last version(~4.2.2), and it worked in same code.

### To Reproduce
Below line added in EtasBus(can.BusABC) __init__ at 4.3.0. It call EtasBus._apply_filters()
`super().__init__(channel=channel, **kwargs)`

```def _apply_filters(self, filters: Optional[can.typechecking.CanFilters]) -> None:
        if self._oci_filters:
            OCI_RemoveCANFrameFilterEx(self.rxQueue, self._oci_filters, 1)
```

_apply_filters got 'filters' as parameter but self._oci_filters is checked.
self._oci_filters is defined after calling _apply_filters in EtasBus __init__.

### Expected behavior
Im not sure how self._oci_filters will be working in EtasBus. I think self._oci_filters could be defined before EtasBus.__init__

### Additional context

OS and version: Windows 10 x64
Python version: 3.10.8
python-can version: 4.3.0
python-can interface/s (if applicable): etas

zariiii9003, 2023-12-05:
@lumagi You have ETAS interfaces, right? Could you try the fix?

lumagi, 2023-12-05:
Yes, I can try the fix, probably tomorrow. Do I understand correctly, this is a regression from the 4.3.0 release?

commit 033be12a (2023-12-08): Fix initialization order in EtasBus (#1704)

super.__init__ calls set_filters, which is only permissible
after the corresponding structures have been created in the child
constructor. Hence, super.__init__ must be called after the child has
finished initialization. Fixes #1693
closed 2023-12-08 by zariiii9003