hardbyte/python-can #1694: Installation instructions for neovi seem to contain an error

grant-allan-ctct, 2023-11-20:
### Describe the bug
On [instructions for installing neoVI](https://python-can.readthedocs.io/en/stable/interfaces/neovi.html) there is a bullet item `Install python-can with the neovi extras` but the incantations given for doing so, in the box below that, seem to be incorrect.  

In ```pip install python-ics[neovi]```  I suspect that the "python-ics" should be replaced by "python-can".

### To Reproduce
To try out my theory, I tried invoking ```pip install python-can[neovi]``` instead of the incantations given in the documentation, and I noticed fun stuff being installed.  Although I am just a newbie to this library, it _seemed_ to me like it succeeded.

But when I instead typed the documented incantations ```pip install python-ics[neovi]``` I was told that my python-ics requirement was already satisfied, and also received additional ```WARNING: python-ics 912.4.post1 does not provide the extra 'neovi'```

Both these behaviors lead me to think that the instruction in the documentation may have an error.

### Expected behavior

### Additional context

OS and version:  Windows10
Python version: 3.11.3
python-can version: 4.3.0
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

zariiii9003, 2023-11-20:
> In `pip install python-ics[neovi]` I suspect that the "python-ics" should be replaced by "python-can".

That's correct, would you like to submit a pull request?

grant-allan-ctct, 2023-11-20:
Yes, thank you, I'd like to give that a shot.  It will be my first one, so I'll leave it until the evening and then see if I can figure out how to proceed with that.

zariiii9003, 2023-11-28:
Resolved in #1697

closed 2023-11-28 by zariiii9003