hardbyte/python-can #1708: The method set_filters doesn’t work sometimes

Cecilia979, 2023-12-22:
### Describe the bug
I used the method set_filters to receive the messages of my specified ID，but the first few messages were not what I expected.

### To Reproduce

    with can.interface.Bus(bustype='vector', channel=2, bitrate=500000,data_bitrate=2000000,app_name="CANoe",fd=True,) as bus:
        bus.set_filters([{"can_id": 0x1b1, "can_mask": 0x7FF, "extended": False}])
        while True:
            msg = bus.recv()
            print("recv",msg)
0x1b1 was the only ID that I expect，but I got others such as 0x3ea，0x335 at the beginning.

### Expected behavior
I expect to receive the specified messages

### Additional context

OS and version: Win10
Python version: 3.8
python-can version:4.3.1
python-can interface/s (if applicable):
[image]
Vector / canoe

hartkopp, 2023-12-22:
Does it generally look like that the interface is enabled BEFORE the filters are applied?

grant-allan-ctct, 2023-12-22:
Could it be related to #1413 and a general problem of receive buffers not necessarily being flushed when a filter is applied?

Cecilia979, 2023-12-26:
> Does it generally look like that the interface is enabled BEFORE the filters are applied?

It seems to be the problem! Because generally speaking, only the first few messages will not be what I expected, while the rest are expected

Cecilia979, 2023-12-26:
> Could it be related to #1413 and a general problem of receive buffers not necessarily being flushed when a filter is applied?

I 我看了这个1413问题，确实是相似的，但是他用的设备是1，而我用的是2，并且我在源码里并没有找到flush_rx_buffer函数，只有flush_tx_buffer，所以实际上我还是不太明白应该怎么解决这个问题。
I have read this 1413 issue, and it is indeed similar. But in this issue the device is kvaser, while I am using vector(appname canoe). And I didn't find the flush_rx_buffer function in the source code. There is only flush_tx_buffer. So in fact, I still don't quite understand how to solve this problem.

PR #1796 (2024-06-21): Activate channels after CAN filters were applied
Fix #1413 
Fix #1708
closed 2024-06-21 by zariiii9003