hardbyte/python-can #1711: can.logger, can.player, and can.viewer fail to initialize gs_usb interface

crashkopf, 2023-12-23:
### Describe the bug

None of the scripts work with gs_usb backend.  All the scripts require specifying interface, channel, and bitrate.  The GsUsbBus class doesn't use channel to find a device, it uses either index or bus and address.  None of these parameters exist on the BusABC class.

### To Reproduce

```command
python -m can.logger -i gs_usb -c whatever -b 500000
```
### Expected behavior

It should "just work".

### Additional context

OS and version:  Windows 10
Python version:  3.10
python-can version:  4.3.1 (commit f0634d2)
python-can interface/s (if applicable): gs_usb

<details><summary>Traceback and logs</summary>

```
python -m can.logger -i gs_usb -c whatever -b 500000
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\me\Documents\GitHub\python-can\can\logger.py", line 247, in <module>
    main()
  File "C:\Users\me\Documents\GitHub\python-can\can\logger.py", line 209, in main
    bus = _create_bus(results, can_filters=_parse_filters(results), **additional_config)
  File "C:\Users\me\Documents\GitHub\python-can\can\logger.py", line 104, in _create_bus
    return Bus(parsed_args.channel, **config)
  File "C:\Users\me\Documents\GitHub\python-can\can\util.py", line 378, in wrapper
    return f(*args, **kwargs)
  File "C:\Users\me\Documents\GitHub\python-can\can\interface.py", line 135, in Bus
    bus = cls(channel, **kwargs)
  File "C:\Users\me\Documents\GitHub\python-can\can\interfaces\gs_usb.py", line 51, in __init__
    raise CanInitializationError(f"Cannot find device {channel}")
can.exceptions.CanInitializationError: Cannot find device whatever
GsUsbBus was not properly shut down
```

</details>

zariiii9003, 2023-12-29:
You should be able to pass keyword arguments to the bus constructor:

```shell
python -m can.logger -i gs_usb -c whatever -b 500000 --index=0
```

crashkopf, 2023-12-29:
Ah, yes, I see.  And it's even stated in the docs under "extra_args", just easy to miss.  I think I was too focused on the --channel argument, which still seems superfluous in this case.  Much appreciated!

closed 2023-12-29 by crashkopf