hardbyte/python-can #1712: I/O Operation on closed file

pipipipi2002, 2023-12-25:
### Describe the bug
I am creating a script to asynchronously receive message and push it to a buffer which will be read by another async method. This receive method will continue running until a CTRL C is detected and shutdown the notifier stop the program.

My implementation is as follows
```python
async def packetConstructor(bus, reader):
    while (True):
        try: 
        # wait for reader to have message
            msg = await asyncio.wait_for(reader.get_message(), timeout=1.0)
            # Push msg to buffer
        except asyncio.TimeoutError:
            pass # prevent await blocking

async def main():
    with can.Bus(interface='canine', bitrate=1000000) as bus:
        reader = can.AsyncBufferedReader()
        logger = can.Logger("canMsg.log")
        listeners: list[MessageRecipient] = [
            reader,
            logger
        ]
        notifier = can.Notifier(bus, listeners, loop=asyncio.get_running_loop())
            
        try:
            packetConstructor_task = asyncio.create_task(packetConstructor(bus, reader))
            await packetConstructor_task
        except asyncio.CancelledError:
            log.debug("Shutting down notifier")
        finally:
            
            notifier.stop()

 
def handler(signal_received, frame):
    log.critical("CTRL-c detected")
    for task in asyncio.all_tasks():
        task.cancel()

if __name__ == "__main__":
    log = get_logger("fw_log")
    signal(SIGINT, handler)
    asyncio.run(main())
```
When I am receiving data and abruptly press Ctrl-c, it will show 
```bash
2023-12-25 23:05:31,162 — fw_log — DEBUG — Received invalid ID
2023-12-25 23:05:31,324 — fw_log — DEBUG — Received invalid ID
2023-12-25 23:05:31,454 — fw_log — DEBUG — Received invalid ID
^C2023-12-25 23:05:31,533 — fw_log — CRITICAL — CTRL-c detected
2023-12-25 23:05:31,667 — fw_log — DEBUG — Shutting down notifier
Exception in callback Notifier._on_message_received(can.Message(t..., 0xff, 0xff]))
handle: <Handle Notifier._on_message_received(can.Message(t..., 0xff, 0xff]))>
Traceback (most recent call last):
  File "/Users/marvinpranajaya/opt/anaconda3/envs/can-testing/lib/python3.12/asyncio/events.py", line 84, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/marvinpranajaya/opt/anaconda3/envs/can-testing/lib/python3.12/site-packages/can/notifier.py", line 146, in _on_message_received
    res = callback(msg)
          ^^^^^^^^^^^^^
  File "/Users/marvinpranajaya/opt/anaconda3/envs/can-testing/lib/python3.12/site-packages/can/listener.py", line 43, in __call__
    self.on_message_received(msg)
  File "/Users/marvinpranajaya/opt/anaconda3/envs/can-testing/lib/python3.12/site-packages/can/io/canutils.py", line 197, in on_message_received
    self.file.write(framestr)
ValueError: I/O operation on closed file.
CANineBus was not properly shut down
```

and the received can message will not be saved in the `canMsg.log` file

### Additional context

OS and version: macos ventura
Python version: 3.12.0
python-can version: 4.2.2
python-can interface/s (if applicable):

Tbruno25, 2024-01-03:
@pipipipi2002 thanks for reporting! Could you please edit your code example to include all imports / dependencies so that the issue can be easily reproduced?

pipipipi2002, 2024-01-03:
Hi @Tbruno25 , Thank you for reaching out. Ive updated the imports in the code above.

closed 2026-11-11 by zariiii9003