hardbyte/python-can #1729: Multiple CAN message thread sending the cycle time is not reliable

xRowe, 2024-01-18:
Hello Developer

I am try to use the Lib to period sending message, now when I periodic sending about 35 CAN message with different ID and different sending cycle. And there is already 42 thread. The sending period become not reliable.

Is there any tricky way to solved the issue? 

For now I am try to send the same cycle messages in a thread. Hope it could work better

j-c-cook, 2024-02-15:
Could you please provide more information about your setup? The following information is minimally required to give any meaningful feedback about the statistics you've shown above:
- hardware interface
- python-can version
- python script

xRowe, 2024-02-16:
@j-c-cook , Thanks for your reply.

Laptop: i5-1145G7 with 8.00 GB RAM
Python: v3.11.7
Hardware interface: Vector
python-can: v4.3.1
Example Script:
```
# Initial Channels
can_bus0 = can.interface.Bus(bustype='vector', channel=0, bitrate=500_000, fd=True, data_bitrate=2_000_000, bitrate_switch=True, receive_own_messages=True, app_name=None)
can_bus1 = can.interface.Bus(bustype='vector', channel=1, bitrate=500_000, fd=True, data_bitrate=2_000_000, bitrate_switch=True, receive_own_messages=True, app_name=None)
can_bus2 = can.interface.Bus(bustype='vector', channel=2, bitrate=500_000, fd=True, data_bitrate=2_000_000, bitrate_switch=True, receive_own_messages=True, app_name=None)
can_bus3 = can.interface.Bus(bustype='vector', channel=3, bitrate=500_000, fd=True, data_bitrate=2_000_000, bitrate_switch=True, receive_own_messages=True, app_name=None)

# Initial Listener for Each Channel for receiving message.
# Listener do some parse job.
_notifier0 = can.Notifier(can_bus0, [listener0], timeout=0.1)
_notifier1 = can.Notifier(can_bus1, [listener1], timeout=0.1)
_notifier2 = can.Notifier(can_bus2, [listener2], timeout=0.1)
_notifier3 = can.Notifier(can_bus3, [listener3], timeout=0.1)

# Start sending message period
can_bus0.send_period("A_2ms", 2/1000.0)
can_bus0.send_period("B_5ms", 5/1000.0)
can_bus0.send_period("C_10ms", 10/1000.0)
can_bus0.send_period("C_20ms", 20/1000.0)
can_bus0.send_period("C_50ms", 50/1000.0)
can_bus0.send_period("C_100ms", 100/1000.0)
can_bus0.send_period("C_200ms", 200/1000.0)

can_bus1.send_period("A_2ms", 2/1000.0)
can_bus1.send_period("B_5ms", 5/1000.0)
can_bus1.send_period("C_10ms", 10/1000.0)
can_bus1.send_period("C_20ms", 20/1000.0)
can_bus1.send_period("C_50ms", 50/1000.0)
can_bus1.send_period("C_100ms", 100/1000.0)
can_bus1.send_period("C_200ms", 200/1000.0)

can_bus2.send_period("A_2ms", 2/1000.0)
can_bus2.send_period("B_5ms", 5/1000.0)
can_bus2.send_period("C_10ms", 10/1000.0)
can_bus2.send_period("C_20ms", 20/1000.0)
can_bus2.send_period("C_50ms", 50/1000.0)
can_bus2.send_period("C_100ms", 100/1000.0)
can_bus2.send_period("C_200ms", 200/1000.0)

can_bus3.send_period("A_2ms", 2/1000.0)
can_bus3.send_period("B_5ms", 5/1000.0)
can_bus3.send_period("C_10ms", 10/1000.0)
can_bus3.send_period("C_20ms", 20/1000.0)
can_bus3.send_period("C_50ms", 50/1000.0)
can_bus3.send_period("C_100ms", 100/1000.0)
can_bus3.send_period("C_200ms", 200/1000.0)

time.sleep(10)

# In total 50+ thread is used for sending/receving/parsing message
```

In the previous snapshot, (is one of the 50ms sending message task) the sending timing monitor by CANoe.

I think the cyclic task is limited by python GIL, and I found that 50+ threading running  the message period sendt with > 100ms task, its period is stable. but 2ms/5ms/10ms/20ms/50ms task/message, it works bad.
And if there is only few threads like 5, the 2ms/5ms/10ms/20ms/50ms task/message is much more stable.

Do you have any advice, what I can do in next step?

Thanks,
Rowe

Enginerd0, 2024-06-28:
Was there any solution to this? I've also had similar issues