hardbyte/python-can #1736: Recive manipulate and send

JatBcn, 2024-01-23:
### Describe the bug
 I am simulating that a signal is coming to the rpi so that I can capture that message edit the message ID and then send it over another socket on my 2-CH CAN FD HAT.

### To Reproduce

In one terminal I execute this python script:
```python
import can
import os
import time

os.system("sudo /sbin/ip link del dev vcan2 type vcan")  # Para el bucle
os.system("sudo /sbin/ip link add dev vcan2 type vcan")  # vcan0 logger
os.system("sudo /sbin/ip link set can0 down")
os.system("sudo /sbin/ip link set can1 down")
os.system("sudo /sbin/ip link set can2 down")
os.system("sudo /sbin/ip link set can0 up type can bitrate 500000")
os.system("sudo /sbin/ip link set can1 up type can bitrate 500000")
os.system("sudo /sbin/ip link set can2 up type can bitrate 500000")
os.system("sudo /sbin/ip link set vcan2 up")

a = 1
bus0 = can.Bus(bustype='socketcan', channel='can0',bitrate = 500000)
bus1 = can.Bus(bustype='socketcan', channel='can1',bitrate = 250000)
bus2 = can.Bus(bustype='socketcan', channel='can2',bitrate = 500000)
busV2 = can.Bus(bustype='socketcan', channel='vcan2',bitrate = 500000)

while True:
	msg = busV2.recv()
	print(msg)
	print(hex(msg.arbitration_id))
	print(msg.data.hex())
	print(msg.channel)
	
	b = hex(msg.arbitration_id + a)
	msg.arbitration_id = int(b,16)
	msg.channel = 'can1'
	print(msg)
	bus1.send(msg)
```
in the other terminal I execute this:
```
@raspberrypi:~ $ cat /home/jlab/Desktop/SMART1_SIMONE-2024-01-16_222654.log  | canplayer vcan2=can0
```

### Expected behavior

signals going out on can1
### Additional context

OS and version:
Python version:
python-can version:
python-can interface/s (if applicable):

<details><summary>Traceback and logs</summary>

</details>

grant-allan-ctct, 2024-01-23:
i see that there is a baudrate mismatch between these two lines.  could it be the trouble?
`os.system("sudo /sbin/ip link set can1 up type can bitrate 500000")`

`bus1 = can.Bus(bustype='socketcan', channel='can1',bitrate = 250000)`

JatBcn, 2024-01-25:
I will try but I think this only affects in de velocity of the output signals, and the problem is that no signals are getting out. 

Thanks for reply!!