hardbyte/python-can #1739: In canfd, when data length > 8, dlc != len(data) in Message._check will cause an error

17ayst, 2024-02-07:
### Describe the bug

when i send message：
can.Message(is_remote_frame=False, is_extended_id=False, dlc=9, arbitration_id=0x22, is_fd=True,
data=(c_ubyte * 12)(*[1]*12), timestamp=time.time(), check=True)
raise ValueError("the DLC and the length of the data must match up for non remote frames"）

### To Reproduce

import can
from ctypes import *
import time
message = can.Message(is_remote_frame=False, is_extended_id=False, dlc=3, arbitration_id=0x22,
                       is_fd=True,
                       data=(c_ubyte * 12)(*[1]*12), timestamp=time.time(), check=True)

### Expected behavior

```
if self.dlc != [0, 1, 2, 3, 4, 5, 6, 7, 8, 12, 16, 20, 24, 32, 48, 64].index(len(self.data)):
    raise ValueError(
                "the DLC and the length of the data must match up for non remote frames"
            )
```

### Additional context

OS and version: windows 10
Python version: 3.7.3
python-can version: 4.2.2
python-can interface/s (if applicable): None

[image]

<details><summary>Traceback and logs</summary>

Traceback (most recent call last):
  File "E:\zhujie\BMAT\bt\c1.py", line 6, in <module>
    data=(c_ubyte * 12)(*[1]*12), timestamp=time.time(), check=True)
  File "E:\zhujie\bmatenv_37\lib\site-packages\can\message.py", line 108, in __init__
    self._check()
  File "E:\zhujie\bmatenv_37\lib\site-packages\can\message.py", line 279, in _check
    "the DLC and the length of the data must match up for non remote frames"
ValueError: the DLC and the length of the data must match up for non remote frames

zariiii9003, 2024-02-07:
The attribute name is misleading, but `can.Message.dlc` is indeed just the data byte count.

17ayst, 2024-02-08:
Thank you very much for your reply
In the canfd frame structure, dlc occupies 4 bytes and can only represent 0-15. 
In canoe, where dlc is selected, the options are also 0-15.
Now, I am writing interfaces based on python-can for several of my CAN devices, but I have encountered this trouble. 
I really hope you can give me some answers. If possible, I am also willing to pull request.

hartkopp, 2024-02-08:
You have to distinguish between the DLC (data length CODE) and the data length itself.
In the physical Classical CAN (CAN CC) and CAN FD frames both DLCs have 4 bit (not 4 byte!!) on the CAN bus bitstream.
But in the case of CAN FD the DLC leads to a non-linear length, see slide 9 of this PDF:
https://github.com/linux-can/can-doc/blob/master/presentations/CAN-XL-Intro.pdf

As @zariiii9003 stated the `can.Message.dlc` data element contains the plain real length information and NOT the DLC. That's why `can.Message.dlc`  is a misleading name for the real length information.

17ayst, 2024-02-09:
Ok, I get it now, thanks！

closed 2024-02-09 by 17ayst