hardbyte/python-can #1742: Detect available configs fails for vector on Windows 11

j-c-cook, 2024-02-14:
### Describe the bug

The `vxlapi` DLL does not appear to be installed using the current [XL Driver Library](https://www.vector.com/int/en/products/products-a-z/libraries-drivers/xl-driver-library/#c75493) (v23.20.2) installer for Windows 11. I've searched both `C:\Windows\SysWOW64` and `C:\Windows\System32` for the `xlvapi` DLL. I've installed many more device drivers than the VN1630A to see if the `vxlapi` DLL would show up. The VN1630A mounts to CANalyzer just fine. I'm thinking this issue may be on Vector's end, but I'm not sure how to escalate the issue if that is the case. 

### To Reproduce

```
import can

available_interfaces = can.detect_available_configs(
        interfaces=['vector']
    )
print(available_interfaces)
```

Output: 
```
Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.
[]
```

### Additional context

OS and version: Windows 11 22H2
Python version: 3.9.18
python-can version: 4.3.1
python-can interface/s (if applicable): vector

zariiii9003, 2024-02-16:
I do have the vxlapi64.dll in my `C:\Windows\System32` directory and the vxlapi.dll in `C:\Windows\SysWOW64`. But i don't if they were put there by the Vector Driver Setup, the XL Driver Library or even by CANape. 🤷‍♂️

Anyhow, i recently pushed a change to use `ctypes.util.find_library` to find the dll, maybe that helps. (#1731)
Otherwise you'd have to manually copy the DLL somewhere on system path.

j-c-cook, 2024-02-16:
I just went and grabbed the `vxlapi` files from a Windows 10 boot and copied them into their respective locations. I can now use my VN1630 as long as CANalyzer has the bus mounted. 

#1731 is in main, and I have tested that for you. 
```
pip uninstall python-can
pip install git+https://github.com/hardbyte/python-can.git@main
```

Both the `v4.3.1` release and main work as long as `vxlapi` is available. I think the missing files resulted in an oversight on Vector's side of things. I'll hope someone who knows a dev there sees this message chain.

closed 2024-02-16 by j-c-cook