hardbyte/python-can #1747: gs_usb does not support set_bitrate() for some Candlelight devices

tuwuhs, 2024-02-26:
### Describe the bug

Cannot set bitrate for some Candlelight devices. I'm using [MKS CANable V2.0 Pro board](https://github.com/makerbase-mks/CANable-MKS) with Candlelight firmware.

### To Reproduce

Demonstrated here on an interactive session:
```python
>>> import usb
>>> import can
>>> dev = usb.core.find(idVendor=0x1d50, idProduct=0x606f)
>>> bus = can.Bus(interface='gs_usb', channel=dev.product, bus=dev.bus, address=dev.address, bitrate=1000000)
>>> bus.gs_usb.set_bitrate(1000000)
False
```

### Expected behavior

`bus.gs_usb.set_bitrate(1000000)` returns `True`.

### Additional context

OS and version: Windows 10
Python version: 3.9.15
python-can version: main (@d40915ca)
python-can interface/s (if applicable):

The board works perfectly fine in Linux / SocketCAN, bitrate can be set through `ip link` command as is usual with native SocketCAN devices.

<details><summary>Traceback and logs</summary>

```python
def func():
    return "hello, world!"
```
</details>

tuwuhs, 2024-02-26:
Checking deeper, looks like the culprit is this: https://github.com/jxltom/gs_usb/blob/master/gs_usb/gs_usb.py#L79

The gs_usb module only implements bit timing conversion for 48 MHz and 80 MHz clocks, while my board apparently runs at 170 MHz.
```python
>>> import usb
>>> import can
>>> dev = usb.core.find(idVendor=0x1d50, idProduct=0x606f)
>>> bus = can.Bus(interface='gs_usb', channel=dev.product, bus=dev.bus, address=dev.address, bitrate=1000000)
>>> bus.gs_usb.device_capability.fclk_can
170000000
```

I think instead of relying on this conversion, we can use the python-can's `BitTiming`. I'll write some quick fix and send a pull request.

zariiii9003, 2024-06-22:
fixed in #1748

closed 2024-06-22 by zariiii9003