hardbyte/python-can #1750: Not able to send CANFD message in "pcan" interface

SachinEngibrains, 2024-03-05:
I am trying to send a CANFD message with a PCAN interface with PCAN USB-FD. I am not getting that sent message on my HW. I am using the below script to send messages.

import time
import can

def initialize_can_bus(channel, bitrate):
    try:
        # Initialize the CAN interface
        can_interface = 'pcan'
        can_bus = can.interface.Bus(channel=channel, bustype=can_interface, bitrate=bitrate,
        fd=True,
        f_clock_mhz=40,nom_brp = 5, nom_tseg1 = 11, nom_tseg2 = 4, nom_sjw = 4, data_brp = 4, data_tseg1 = 3, data_tseg2 = 1, data_sjw = 1)

        # Log initialization details
        print(f"CAN bus initialized successfully on channel {channel} with bitrate {bitrate}.")

        return can_bus

    except can.CanError as e:
        print(f"Error initializing CAN bus on channel {channel} with bitrate {bitrate}: {e}")
        return None

# Main script
if __name__ == "__main__":
    channel = 'PCAN_USBBUS1'
    bitrates = [2000000]  # 500k, 1M, and 2M bitrates

    for bitrate in bitrates:
        print(f"Attempting to initialize CAN bus on channel {channel} with bitrate {bitrate}...")
        can_bus = initialize_can_bus(channel, bitrate)

        if can_bus is not None:
            # Prepare a CAN message
            can_message = can.Message(arbitration_id=0x600, data=[0x22, 0x22, 0, 0, 0, 0, 0, 0], is_extended_id=False, is_fd=True,bitrate_switch=True)

            # Send the CAN message
            try:
                can_bus.send(can_message)
                print("CAN-FD message sent successfully.")

            except can.CanError as e:
                print(f"Error sending CAN-FD message: {e}")

            # Allow time for message transmission
            time.sleep(1)

            # Close the CAN bus
            can_bus.shutdown()
            print("CAN bus shutdown.\n")

Output:
Attempting to initialize CAN bus on channel PCAN_USBBUS1 with bitrate 2000000...
uptime library not available, timestamps are relative to boot time and not to Epoch UTC
CAN bus initialized successfully on channel PCAN_USBBUS1 with bitrate 2000000.
CAN-FD message sent successfully.
CAN bus shutdown.

But not receive this message in my HW and red LED is blinking in PCAN USB-FD device.
### Additional context

OS and version: Windows 10
Python version: Python 3.9.0
python-can version: python-can 4.3.1
python-can interface/s (if applicable): pcan

btRooke, 2024-04-08:
Hi - I've been working with this library and a PCAN FD USB interface with no problems.

Setup:
- Windows 11
- Python 3.9.4
- python-can 4.3.1
- PCAN-FD USB dongle

My bit timings (for 1MHz arbitration and 2MHz data rate) are:

```ini
f_clock_mhz = 80

nom_brp = 4
nom_tseg1 = 14
nom_tseg2 = 5
nom_sjw = 4

data_brp = 2
data_tseg1 = 15
data_tseg2 = 4
data_sjw = 4
```

Potentially the PCAN won't accept such few time quanta with your timings (3 + 1 + 1 = 5)?

`data_brp  = 1`, `data_tseg1 = 14`, and `data_tseg2 = 5` might be worth a shot...

Cesare12, 2024-04-18:
Are you using virtual environment?

LenaJok, 2024-07-09:
Hi, I am having the same issue. I am using exactly the same settings as seen in PCAN View, in my script, but I only get error frames while using PCAN View I can send and receive messages normally. 

Can somebody please assist?

My PCAnBasic.dll is 4.8.2
Python version: Python 3.10.5
python-can version: python-can 4.3.1
python-can interface/s (if applicable): pcan

Cesare12, 2024-07-10:
try use Python 3.7-32

funnyrabbitvu, 2024-08-29:
> Hi, I am having the same issue. I am using exactly the same settings as seen in PCAN View, in my script, but I only get error frames while using PCAN View I can send and receive messages normally. 
> 
> 
> 
> Can somebody please assist?
> 
> 
> 
> PCAN-FD USB dongle
> 
> PCANBasic.dll is 4.8.2
> 
> Python version: Python 3.10.5
> 
> python-can version: python-can 4.4.2 (and other versions as well)
> 
> python-can interface/s (if applicable): pcan

Hi, did you fix it yet? Same issue here.