hardbyte/python-can #1761: Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies).

Dr-Br0t, 2024-03-27:
### Describe the bug

I have been working with this library on Windows 10 for some months, I am trying to run some scripts into my new Windows 11 computer, but when I run the script appears a error with this description: "Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax."

I don not remember if I had to do an special step aside installing the Vector XL drivers. I will appreciate any kind of help.

### To Reproduce
Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.

### Expected behavior
The script should run

### Additional context

OS and version: Windows 10, python 3.11
Python version: 3.11
python-can version: 4.3.1
python-can interface/s (if applicable): Vector

Code:
from can import Notifier, SizedRotatingLogger
from can.interfaces.vector import VectorBus

Error:
Could not import vxlapi: Could not find module 'vxlapi64' (or one of its dependencies). Try using the full path with constructor syntax.

FedericoSpada, 2024-04-02:
Same problem on Windows 11.
I've fixed it by installing XL Driver Library and manually coping the new file "C:\Users\Public\Documents\Vector\XL Driver Library 20.30.14\bin\vxlapi64.dll" into "C:\Windows\System32", as reported here:
https://stackoverflow.com/questions/74755120/could-not-import-vxlapi-could-not-find-module-vxlapi64

Dr-Br0t, 2024-06-11:
It works

CN-CaoJun, 2025-06-16:
it works too.