hardbyte/python-can #1770: Listener.stop() method breaks API since declared abstract

acolomb, 2024-04-24:
In #1724, the `can.listener.Listener.stop()` method was decorated as an `@abstractmethod`, without any further explanation.  This seems to work alright within the python-can codebase, but other dependent projects will get in trouble if they don't implement that method on derived classes.  At the very least should this API change be documented prominently.

This actually causes trouble with e.g. https://github.com/christiansandberg/canopen/, see https://github.com/christiansandberg/canopen/blob/713f005af8a6787a4ff54a26f55fa6736f86abaf/canopen/network.py#L346

### Expected behavior
This API change should be clearly documented OR reverted, going back to the empty default implementation.

### Additional context

OS and version: Linux Ubuntu 22.04
Python version: 3.11.6
python-can version: 4.0.0-dev.2-628-g7dba4490 (current main branch)
python-can interface/s (if applicable): does not matter

<details><summary>Example script to reproduce</summary>

```python
import can

class MessageListener(can.Listener):

    def on_message_received(self, msg):
        pass

MessageListener()
# TypeError: Can't instantiate abstract class MessageListener with abstract method stop
```
</details>

zariiii9003, 2024-04-24:
You are correct, it must be mentioned in the changelog #1759. Another reason not to release it as 4.3.2, but 4.4.0.
Thanks for testing the pre-release!

acolomb, 2024-04-24:
You're welcome.  I actually only switched to the main branch for investigating a fix for #1771.

Will make a PR in the canopen package to deal with this change, to hopefully stay compatible right after your v4.4.0 release.

PR #1795 (2024-06-12): Remove abstractmethod() decorator from Listener.stop()
Closes #1770 

This is breaking some downstream code.
closed 2024-06-12 by zariiii9003